chipsalliance/verible #2035: Lint configuration rules on nvim

msebgarcia, 2023-11-01:
Hello,

I'm trying to disable some lint rules and enable others but I'm not hitting the right key; I installed it through Mason and tried different stuff (mostly related to config from other linters) but nothing worked.

Any help would be greatly appreciated, thanks in advance.

IEncinas10, 2023-11-04:
Hello, as you weren't too specific I will be a bit verbose. Feel free to ask again if this doesn't help you.

If you haven't, check:

- https://github.com/chipsalliance/verible/blob/master/verilog/tools/ls/README.md#neovim

- https://github.com/chipsalliance/verible/blob/master/verilog/tools/ls/README.md#configuring-language-server-arguments

My config (using the one in the examples as a reference, only showing the changed part):

```lua
local lspconfig = require("lspconfig")

lspconfig.verible.setup {
    on_attach = on_attach,
    cmd = {
      "verible-verilog-ls",
      "--rules_config_search",
    },
    root_dir = function() return vim.loop.cwd() end
}
```

You can customize the lsp arguments by adding lines to `cmd`. In the link above you have some arguments expained, but to get full descriptions you can do `verible-verilog-ls --helpfull`.

If you want to use the same config always, you can craft a configuration file, put it somewhere and write the `cmd` like this:

```lua
    cmd = {
      "verible-verilog-ls",
      "--rules_config=/FULLPATH/TO/HARDCODED_VERIBLE_RULES",
    },

```

or you can specify the configuration via arguments

```lua
    cmd = {
      "verible-verilog-ls",
      "--rules=-module-filename", -- verible-verilog-lint --help_rules all
    },

```

### Debugging LSP

To debug LSP issues, `vim.lsp.set_log_level("debug")` and then doing `:LSPInfo` from NVIM will be very helpful.

After you're done debugging, I would `vim.lsp.set_log_level("off"), as this file can get very big. 

Reference: https://github.com/neovim/nvim-lspconfig#troubleshooting

msebgarcia, 2023-11-04:
Hi! That was more that what I needed, thanks a lot. I was so blind about to nvim side that that was all I look for, never crossed my mind to check non lint related docs and my searchs outside repo where biased by the same mistake.

Thank you a lot!

closed 2023-11-04 by msebgarcia