chipsalliance/verible #2041: Insufficient forbidden-macro documentation

matlupi, 2023-12-06:
**Describe the bug**

Description of `[forbidden-macro]` is insufficient.

**Actual vs. expected behavior**

"Checks that no forbidden macro calls are used."

It is not clear what is a forbidden macro and what is not.

E.g. of resolution: "Checks that none of the following forbidden macros are used (uvm_warning,...)"

gobbedy, 2025-01-21:
Hello, could someone please explain why `uvm_warning is forbidden? I checked the verible linter documentation and searched the verible github issues and could not find any explanation. I also googled and found no discussions of `uvm_warning being broken in any way..

matlupi, 2025-01-22:
> Hello, could someone please explain why the uvm_warning macro is forbidden? I checked the verible linter documentation and searched the verible github issues and could not find any explanation. I also googled and found no discussions of `uvm_warning being broken in any way..

@gobbedy, I believe the reason for that is that a warning does automatically lead to a failure in the test.
Meaning that most of the automatic checks (e.g. when regression testing) will not show any failure if warnings are fired.
Personally I like to use warning as a way to debug, meaning, if in a UVM test something worth of attention happens, I flag it with a UVM_WARNING. If the test passes, no one will ever look at the logs. If the test fail it is a first clue to start looking from. Clearly depending on the size of the verification team this can easily get out of hand.

gobbedy, 2025-01-30:
@matlupi thank you, that makes sense. We have a parser that highlights all warnings clearly at the end so they are not missed and we have an explicit end-of-project phase where we must ensure that all testcases pass without warnings (valid warnings can be waived).

Makes sense that some teams would forbid it. As long as it remains possible in the future to exclude `uvm_warning from the 'forbidden' list as forbidden-macro expands to other macros..