chipsalliance/verible #2053: [Help or suggestion] How to disable short if-else statements on a single line?

beyond-fu, 2023-12-18:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
always_ff @(posedge clk, posedge rst) begin
    if (rst) dout <= 2'b0;
    else if (en1) dout <= 2'b01;
    else if (en2) dout <= 2'b10;
end
```
**Actual output**

```systemverilog
// This doesn't look right.
always_ff @(posedge clk, posedge rst) begin
    if (rst) dout <= 2'b0;
    else if (en1) dout <= 2'b01;
    else if (en2) dout <= 2'b10;
end
```
**Expected or suggested output**

```systemverilog
// This result would look better from the formatter.
always_ff @(posedge clk, posedge rst) begin
    if (rst) 
        dout <= 2'b0;
    else if (en1) 
        dout <= 2'b01;
    else if (en2) 
        dout <= 2'b10;
end
```
So is there a way to make the if-else statement and assignment statement on the different lines for current release? If not, I strongly strongly suggest to give an option to enable this feature, just like `AllowShortIfStatementsOnASingleLine=Never` option in `clang-format`. I really don't like they are on the same line even if no `begin-end block`!

Thank you!

leosmaia21, 2024-03-13:
Hello,
Did you manage to solve it? I am looking for that also.

beyond-fu, 2024-03-14:
Haha, I am not alone. No, I give up that. BTW, `verible` updates so slow :(

Uneducated-Fly, 2024-06-26:
Hopefully the authors can solve this problem quickly！Thanks！