chipsalliance/verible #2073: Formatter removes newline between successive ` directives

kbrunham-intel, 2024-01-13:
**Test case**

```systemverilog
`timescale 1 ps / 1 ps 
`default_nettype none
```

Include any options or configuration used.

**Actual output**

```systemverilog
`timescale 1 ps / 1 ps `default_nettype none
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

```systemverilog
`timescale 1 ps / 1 ps 
`default_nettype none
```

Citations to published style guides would help.

kbrunham-intel, 2024-01-13:
I'm willing to investigate this issue, please consider assigning it to me.

IEncinas10, 2024-01-13:
Thanks for working on it :) Feel free to ping me if you want issues assigned

kbrunham-intel, 2024-01-14:
@IEncinas10 : I've reviewed the code, and suspect we are not making a partition correctly. Who could I review with to make sure this is the right direction?

IEncinas10, 2024-01-14:
I'm not sure, perhaps @hzeller could tell.

I haven't worked on the formatter but I can **try** to help