chipsalliance/verible #2074: Migrate name style rules to regex

IEncinas10, 2024-01-13:
There are several "name style rules" and one more being developed (https://github.com/chipsalliance/verible/pull/1983) which have their rules hardcoded or with small configuration capabilities.

In the past, this motivated a PR that eventually didn't land but did the same thing (using `std::regex` which seemed to be the problem). The topic was raised again in https://github.com/chipsalliance/verible/pull/2017. 

This is a "macro issue" to track the conversion of the existing rules to be regex-based (using `re2`). I'll keep this in mind and work on it on spare time, I don't know if I'll tackle this on a per-rule bases or I'll group various rules on the same PR.

- [ ] constraint_name_style
- [ ] enum_name_style
- [ ] interface_name_style
- [ ] macro_name_style
- [ ] parameter_name_style
- [ ] parameter_type_name_style
- [ ] signal_name_style
- [ ] struct_union_name_style
- [ ] ¿dff_name_style? (if it lands)

If anyone comes by this issue and wants to help feel free to do so 😃

sconwayaus, 2024-04-05:
Can you add port_name_suffix to the list too. It's basically the same (checks style (snake_case) with some prefixes), and needs some configuration. Regex looks like a good answer.

Happy to with this one too.

IEncinas10, 2024-04-05:
Missed this one, thanks!

sconwayaus, 2024-04-07:
Happy to help out with any of these rules too.

I've already been playing with the port_name_suffix rule before I saw this issue, and added configuration to it. Just need to convert this over to regex (add tests, etc...).

burnfield, 2024-05-05:
Would be awesome. This has been implemented in [svlint](https://github.com/dalance/svlint/blob/master/MANUAL.md), if you're looking for inspiration.