chipsalliance/verible #2076: Rejected valid signal name

matlupi, 2024-01-19:
**Describe the bug**

Short summary.

**To Reproduce**

```bash
verible-verilog-lint test.sv
```

with `test.sv`

```systemverilog
module test;
  logic analog;
endmodule
```

**Actual behavior:**

Rejects valid syntax 

```
test.sv:2 9-14 syntax error at token "analog"
```

**Expected behavior**

No syntax error.
`analog` is not a reserved keyword according to LRM Annex B Table B.1

hzeller, 2024-01-31:
Yeah, this is because we accept a few more [potential keywords from Verilog-AMS](https://github.com/chipsalliance/verible/blob/c548dbb417db726fcf6f06d31496cca2df56bac9/verilog/parser/verilog.lex#L716)

We should probably add that to the 'keyword or identifier' classification like here
https://github.com/chipsalliance/verible/blob/c548dbb417db726fcf6f06d31496cca2df56bac9/verilog/parser/verilog.y#L757-L782

matlupi, 2024-02-01:
Is there a switch to avoid this syntax error?
Since the file is not a VAMS file, I should be able to get around that, right?

matlupi, 2024-02-02:
Will do, thanks!