chipsalliance/verible #2127: How to debug VSCode LS?

shareefj, 2024-03-03:
Using VSCode 1.86.2

**What activity failed**

I can't get the plugin to obey any of the formatter settings.  It looks like it's using the default "infer" state as I can get assignments to align, for example, if I manually insert spaces until it identifies what I'm trying to do.

So what is the correct format of the `verible.arguments` list in `settings.json`?  I've tried the following:

```
  // Verible Formatter
  "verible.path": "/opt/verible/latest/bin/verible-verilog-ls",
  "verible.arguments": [
    "inplace=true",
    "column_limit=100",
    "indentation_spaces=4",
    "assignment_statement_alignment=align",
    "named_parameter_alignment=align",
    "formal_parameters_alignment=align",
    "port_declarations_alignment=align",
    "port_declarations_right_align_unpacked_dimensions=true",
    "port_declarations_right_align_packed_dimensions=true",
  ],
```
I also tried with a space separated list and swapping the equals for a space.  None of which worked nor caused any sort of error/warning in the log.

And obviously running the same commands from the command line `verible-verilog-formatter` does what I want.

**Logfiles**

Here's the output of one of the log files.  I didn't have a `verible.filelist` file initially (and saw errors) so created one but it had no effect.

```
Verible Verilog Language Server built at v0.0-3539-g9442853c
I0303 15:21:07.822356   17324 verilog-language-server.cc:224] Initializing with project-root '/home/shareefj/git/nuv'
I0303 15:21:07.822486   17324 symbol-table-handler.cc:96] Found file list under /home/shareefj/git/nuv/verible.filelist
I0303 15:21:07.827696   17324 json-rpc-dispatcher.cc:70] Ignoring notification 'initialized'
I0303 15:21:18.616288   17324 json-rpc-dispatcher.cc:70] Ignoring notification '$/setTrace'
OK
Shutting down due to shutdown request.
Statistics
Largest message seen: 6 kiB 
    $/setTrace (unhandled)  ev         1
                initialize RPC         1
   initialized (unhandled)  ev         1
                  shutdown RPC         1
      textDocument/didOpen  ev         1
textDocument/documentSymbol RPC         1
```

So could someone define the format expected within `settings.json` and then document it somewhere?  And if that isn't the issue in this case, suggest how I can debug this further?

Thanks.

hzeller, 2024-03-05:
I need to double check that all the formatting paramters are passed through properly.

But for debugging, try to set the following environment variables, they will increase the loglevel.

```
VERIBLE_LOGTHRESHOLD=0 VERIBLE_VLOG_DETAIL=1
```

Don't know if you can set environtment variables directly in VSCode, I usually just set them in the shell I start vscode, then they wil be inherited.

hzeller, 2024-03-05:
I've updated the code to emit the command line at startup of the language server. This should help to see if the command line flags arrive where they should. You should be able to see them and restart vscode after updating the language server.

Note, you don't need to set any environment variables as suggested earlier, it will always show up.

If you watch the output, make sure that all the arguments start with double-dash. So it needs to look like `--indentation_spaces=4` ... maybe that is the problem ?

hzeller, 2024-03-15:
Did it help ?

shareefj, 2024-09-18:
Sorry, I totally missed this reply.

So I have a version from Aug installed that has your update and I can see the command line being output.  The issue does seem to be that the arguments require a prefix of `--`.  Could the language server be updated to moan if it finds arguments that it doesn't recognise?  For example, I can add an argument of `named_port_alignment align` and it happy accepts it but obviously doesn't do anything with it.  I don't think the language server accepts non-flag arguments does it?  So it shouldn't probably just barf if it sees them.

My indentation is now behaving as expected but no matter what I try, I can't change the column limit from the default 100.  Perhaps that option is really broken.  I can get the language server to error if I mis-spell the argument but any change to the value isn't updated even after restarting VSCode.