chipsalliance/verible #2140: Windows CI currently failing; wants chocolatey credentials ?

hzeller, 2024-03-10:
Currently, the CI on Windows is broken:
https://github.com/chipsalliance/verible/actions/runs/8218621248/job/22479126298#step:5:17

Looks like the package manager used there is needing some sort of credentials. Why would credentials be needed to download from the package manager ? Maybe the URL changed ?

hzeller, 2024-03-10:
I have no idea how things work on Windows, paging @corco for expertise.

hzeller, 2024-03-10:
Hold on, maybe it fixed itself ...

corco, 2024-03-10:
The info I found seems to point to a server issue with Chocolatey (or Cloudflare) or an environment issue. I did a PR to update Windows runner to 2022 (from 2019).

Otherwise I would give it a day or two and hope it solves itself.

closed 2024-03-14 by corco