chipsalliance/verible #2145: Unexpected formatting of interface classes

sriyerg, 2024-03-14:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
interface class foo;
       pure virtual function void bar(    );
  pure       virtual task baz(
  );
    endclass
```

Include any options or configuration used.
`--inplace`

**Actual output**

```systemverilog
// This doesn't look right.
  interface class foo; pure virtual
  function void bar()
  ; pure virtual
  task baz()
  ;
  endclass
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

```systemverilog
// This result would look better from the formatter.
interface class foo;
  pure virtual function void bar();
  pure virtual task baz();
endclass
```

Version:
```console
$ verible-verilog-format --version
v0.0-3617-g4f98e145
Commit	2024-03-10
Built	2024-03-11T00:27:27Z
```

Citations to published style guides would help.
I would have expected interface classes to be formatted as a regular class, but the formatting is way off, as shown above.

sriyerg, 2024-03-14:
Looks like this was already reported over 2 years ago in #1149. Nevertheless, requesting renewed attention on this, hoping not too difficult to fix. Feel free to close bug on ack.

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

Same `interface class` smash as #1149; reduced case now formats as expected on this tag.