chipsalliance/verible #2146: VSCode LSP crashes while adding module inputs/outputs

The-MEO, 2024-03-19:
I use VSCode version 1.84.1 with the ssh remote Extension.
On the remote machine the following extensions are installed:
- CHIPSAlliance.verible
- eirikpre.systemverilog

-----

**What activity failed**
While editing the input/output list of a module the language server crashes.

Unfortunately neither disabling the extension _eirikpre.systemverilog_ nor removing all argument flags in _verible.arguments_ fixes this crashes.

**Expectation**
Get only linting errors for syntax error while typing.

**Test case**

```systemverilog
//*****************************************************************************
// (C) Copyright info
//*****************************************************************************

module flag_generator#(
    parameter MEM_ADDR_WIDTH = 8  // number of <data_width> words in memory
) (
    input i_clk,     // input clock
    input i_rst_an,  // asynchronous reset
    input i_rst_sn,  // synchronous reset // crash when adding ,

    input //crash after typing t

);
    
endmodule
```

**Logfiles**

```txt
Verible Verilog Language Server built at v0.0-3500-g31182be1
F0319 08:58:30.832713     918 interval_set.h:638] Check failed: p.second Failed to emplace!
*** Check failure stack trace: ***
    @           0x5eb00d  absl::lts_20230802::log_internal::LogMessageFatal::~LogMessageFatal()
    @           0x4039b4  verilog::VerilogProject::ContentToFileIndex::Register()
    @           0x47f8e6  verilog::VerilogProject::UpdateFileContents()
    @           0x41c6a9  verilog::VerilogLanguageServer::UpdateEditedFileInProject()
    @           0x46e6ec  std::_Function_handler<>::_M_invoke()
    @           0x5d0cf8  verible::lsp::BufferCollection::didChangeEvent()
    @           0x5d215a  std::_Function_handler<>::_M_invoke()
    @           0x5e4dc1  verible::lsp::JsonRpcDispatcher::CallNotification()
    @           0x5e53e2  verible::lsp::JsonRpcDispatcher::DispatchMessage()
    @           0x5e5a73  verible::lsp::MessageStreamSplitter::ProcessContainedMessages()
    @           0x5e5fe9  verible::lsp::MessageStreamSplitter::ReadInput()
    @           0x5e60bd  verible::lsp::MessageStreamSplitter::PullFrom()
    @           0x41c35e  verilog::VerilogLanguageServer::Run()
    @           0x4133b3  main
    @           0x68fada  __libc_start_call_main
*** SIGABRT received at time=1710835110 on cpu 1 ***
PC: @           0x6c77cc  (unknown)  __pthread_kill
    @           0x6a7980  (unknown)  (unknown)
[Error - 8:58:30 AM] Connection to server got closed. Server will not be restarted.
```

_Thank you for your hard work to make verible available._

hzeller, 2024-03-19:
+1 to what @IEncinas10 said. The file update stuff was fixed around that time.

The-MEO, 2024-04-22:
Thank you for your fast response.
In the meantime I had the opportunity to try out the latest version for a few days and it was super stable.
Thank you very much for your efforts.

closed 2024-04-22 by The-MEO