chipsalliance/verible #2148: Generate default .rules.verible_lint from verible-verilog-lint

sconwayaus, 2024-03-22:
**Summary**
Generating a complete '.rules.verible_lint' file is a bit tedious to do by hand. It would be better to be able to get a default  '.rules.verible_lint' file out of the lint tool as a starting point.  

**Test cases and examples**
Execute 'verible-verilog-lint --print_rules_file' to print the default rules to stdout. Users can then pipe/copy&paste the output to a file for latter use ('.rules.verible_lint' for example).

Example output:
~/git/verible/bazel-bin/verilog/tools/lint$ ./verible-verilog-lint --print_rules_file
+always-comb
+always-comb-blocking
+always-ff-non-blocking=catch_modifying_assignments:false;waive_for_locals:false
-banned-declared-name-patterns
+case-missing-default
+constraint-name-style
+create-object-name-match
-disable-statement
-endif-comment
...

**Proposal**
I've implemented the above in a fork, mainly to see if I could implement the feature, with the view to start a pull request. Before I do is there is any appetite for such a feature?

**Additional context**
None

hzeller, 2024-03-22:
I like that, please send a pull request!

IEncinas10, 2024-04-22:
Closing, thanks for the PR!

closed 2024-04-22 by IEncinas10