chipsalliance/verible #2164: Repeated Diagnostic Messages in coc.nvim with Verible Language Server

TBD9rain, 2024-04-18:
My IDE is as following:
- gvim (v9.0.1897) with coc.nvim (v0.0.82-84ce0312)
- Verible Verilog Language Server built at v0.0-3638-ge3ef2a37
-----

# Repeated Warning Messages

The verible-verilog-ls configurations are as following:

```json
"languageserver": {
        "verible":{
            "enable": true,
            "command": "verible-verilog-ls",
            "filetypes": ["verilog", "systemverilog"],
            "trace.server":"messages"
        }
    }
```

When I created a new `.v` file and typed several spaces, repeated identical warning messages popped up:

[image]

I expect a single warning message per actual issue per file change or analysis trigger.
After I configured the language server as following in coc.nvim, the second warning message dissapeared.

```json
"languageserver": {
        "verible":{
            "enable": true,
            "command": "verible-verilog-ls",
            "filetypes": ["verilog", "systemverilog"],
            "disabledFeatures": ["diagnostics"], //  fix repeated identical messages bug
            "trace.server":"messages"
        }
    }
```

[image]

But I don't think this is an expected behavior or a long-term solution.

# Log Messages

Logs of the preceeding different configurations are smilar except the message timestamps and durations.

```txt
[Trace - 22:45:20.639] Sending request 'initialize - (0)'.
Verible Verilog Language Server built at v0.0-3638-ge3ef2a37
commandline: C:/Program Files/Vim/vimfiles/languageserver/verible/verible-verilog-ls.exe 
[Trace - 22:45:20.647] Received response 'initialize - (0)' in 8ms.
[Trace - 22:45:20.647] Sending notification 'initialized'.
[Trace - 22:45:20.650] Sending notification 'workspace/didChangeConfiguration'.
[Trace - 22:45:20.650] Sending notification 'textDocument/didOpen'.
[Trace - 22:45:20.650] Sending request 'textDocument/diagnostic - (1)'.
[Trace - 22:45:20.651] Received notification 'textDocument/publishDiagnostics'.
[Trace - 22:45:20.651] Received response 'textDocument/diagnostic - (1)' in 1ms.
[Trace - 22:45:40.429] Sending notification 'textDocument/didChange'.
[Trace - 22:45:40.429] Sending request 'textDocument/diagnostic - (2)'.
[Trace - 22:45:40.429] Received notification 'textDocument/publishDiagnostics'.
[Trace - 22:45:40.430] Received response 'textDocument/diagnostic - (2)' in 1ms.
[Trace - 22:45:41.191] Sending notification 'textDocument/didChange'.
[Trace - 22:45:41.193] Sending request 'textDocument/diagnostic - (3)'.
[Trace - 22:45:41.194] Received notification 'textDocument/publishDiagnostics'.
[Trace - 22:45:41.195] Received response 'textDocument/diagnostic - (3)' in 2ms.
```

Verbose logs of the two configurations are also similar except the message timestamps and durations.

hzeller, 2024-04-18:
On every change of the file, the diagnostics are updated. It seems like your editor does not coalesce these ?

There are also two mechanisms diagnostics can be received:
  * the default by Verible is to publish new diagnostics whenever some change happens in the file (`textDocument/publishDiagnostics`)
  * It _also_ seems that your editor is requesting diagnostics independently (`textDocument/diagnostic`), so maybe this is where things get merged and show up multiple times ? Is the `"disabledFeatures"` configuration switching the explicit request off ?

So things do look like they work as intended, just that the editor does not remove the duplicates.

I'll have a look if it is possible by the editor to request not auto-publishing diagnostics when it actually does the request by itself (maybe there is a setting the editor can send to the language server).

But  you should probably also file an issue with your editors' language server implementation to make sure they coalesce the identical diagnostic messages properly or ignore diagnostic notifications when they already request

TBD9rain, 2024-04-19:
The `"disabledFeatures"` configuration doesn't switch the explicit request off. I think it just diabled one of the messages from diaplaing.

I will file an issue with the language server implementation plugin as well.

fannheyward, 2024-04-19:
> Is the "disabledFeatures" configuration switching the explicit request off ?

The `disabledFeatures` made coc.nvim ignore publishDiagnostics one.

verible server files the capabilities with `diagnosticProvider`, coc.nvim will use pull mode to request to server for diagnostics.

I think the server should use both publishDiagnostics and diagnosticProvider at the same time.

benjamin051000, 2024-08-02:
Just chiming in that this is also an issue with `nvim-lspconfig`:
[image]

Here's my setup:
```lua
-- ~/.config/nvim/after/plugin/lsp.lua
require'lspconfig'.verible.setup{}
```

I'm not sure what the differences are between the implementation of Verible and other language servers, but this is the only language server that outputs diagnostic messages twice for me.

hzeller, 2024-08-02:
So maybe we need to change the language server to not auto-publish findings whenever the editor is in a mode where it actively pulls the diagnostics regardless if there was an auto-diagnostic already sent.
I am having some vacation coming up, so I might have some time to look at it.

firerock1, 2025-02-10:
Is there any possible way to fix this repeated diagnostics now👀

qarlosalberto, 2025-03-30:
I have the same problem in VSCode:

[image]

qarlosalberto, 2025-03-30:
For VSCode it works fine (duplicate bug) with:

```
"vscode-languageclient": "^7.0.0"
```

But not with:

```
"vscode-languageclient": "^9.0.1",
```

hzeller, 2025-03-30:
In #2379 , I've now disabled pushing diagnostic notifications by default.
If the language server actively pulls diagnostics, they will show up, but the duplicate unsolicited push of diagnostics is not happen anymore, so you won't see duplicates.

Can you check that this solves what you're seeing ?

IEncinas10, 2025-04-05:
I just saw this and I can say this is fixed for me. I had this ugly hack on my nvim config

```lua
  on_init = function(client)
    if client.supports_method('textDocument/publishDiagnostics') then
      vim.lsp.diagnostic.on_publish_diagnostics = function() end
    end
  end,
```

which isn't needed anymore. If I use the `--push_diagnostic_notifications` flag the issue comes back. Thank you!

I'll close the issue. Feel free to reopen.

closed 2025-04-05 by IEncinas10