chipsalliance/verible #2171: automatically comment preprocessor conditionals

fangism, 2024-04-30:
**Summary**

This request captures features from a script that lived outside of Verible, but some developers found useful.
The script (by @jonmayergoogle ) would automatically add comments after `ifdef` `elsif` `else` `endif` directives that reflected the condition under which the section was active, and served as a readability aid.

**Test cases and examples**

```systemverilog
/* Hello, world. */
alpha
`ifdef FOO
/*
beta
`ifndef BAR
*/
luke
`ifdef BUM  // is BUM defined
wookie
`endif  // this comment gets removed
r2d2
`endif
wedge
```

would automatically become annotated like:

```systemverilog
/* Hello, world. */
alpha
`ifdef FOO  // FOO
/*
beta
`ifndef BAR
*/
luke
`ifdef BUM  // FOO && BUM
wookie
`endif  // FOO && BUM
r2d2
`endif  // FOO
wedge
```

**Proposal**

The implementation for this would be based on an unpreprocessed token stream.  No syntactic validation is needed, as this is purely a transformation based on preprocessing tokens.  The simplicity of this task means that we don't need a full syntax tree transformation/rewriting framework.

**Additional context**

cl/629042737 (Google)

kuku929, 2024-09-30:
hello, this seems interesting. In which file should I put the code for the feature?

Bhavin-umatiya, 2024-12-25:
@fangism Hello
please check any further change so u can guide me i'll try

fangism, 2024-12-29:
> @fangism Hello please check any further change so u can guide me i'll try

Got your message.  I'm a little occupied over the holidays but will try to take a look at https://github.com/chipsalliance/verible/pull/2314.