chipsalliance/verible #2173: linter cannot find macro defined in a different file

samimia-swks, 2024-05-04:
I have a mymodule.sv file which uses a simple macro that's defined in another file (say macros.sv). 
`define iw(fp) $bits(fp.iw)` 

calling `verible-verilog-lint mymodule.sv` results in: 
```preprocessing error at token "`iw" : Error expanding macro identifier, might not be defined before.```

As far as I can tell there is no way to pass macros.sv to the linter so it knows where to find the macro. 
But that seems hard to believe since 99% of the time people define their macros all over the place. 

The language server seems to have a way to pass a list of files, but even when that's done, the same error appears via the language server:

[image]

IEncinas10, 2024-05-11:
Could you perhaps share a minimal reproducer and the version of verible you're using?

samimia-swks, 2024-05-11:
ok I think the problem is a bit different than what I described. 
The issue is that if a macro is defined in a seperate file than where it is used, an unrelated syntax error in the file where the macro is used is always blamed on the macro. 

Let me illustrate. 

Say I have an example.sv containing a macro definition and usage. Calling verible-verilog-lint identifies the always-ff-non-blocking  issue on line 4 : 
```
$ cat example.sv 
`define ABS_MACRO(x)         (((x) < 0) ? -(x) : (x))
module example(input clk);
parameter int Y = `ABS_MACRO(-5); // macro line
always_ff @(posedge clk) q = 2;
assign m = 2;
endmodule
$ verible-verilog-lint  example.sv 
example.sv:4:26: Use blocking assignments, at most, for locals inside 'always_ff' sequential blocks. [Style: sequential-logic] [always-ff-non-blocking]
$ verible-verilog-lint --version
v0.0-3644-g6882622d
Commit  2024-04-22
Built   2024-04-23T04:20:31Z
```
if I introduce a syntax error on line 5, it is detected correctly : 
```
$ cat example.sv 
`define ABS_MACRO(x)         (((x) < 0) ? -(x) : (x))
module example(input clk);
parameter int Y = `ABS_MACRO(-5); // macro line
always_ff @(posedge clk) q = 2;
assign m  2;
endmodule
$ verible-verilog-lint  example.sv 
example.sv:5:11: syntax error at token "2"
```
Now I move the macro to a different file and call the tool again passing both files: 
```
$ cat macros.sv 
`define ABS_MACRO(x)         (((x) < 0) ? -(x) : (x))
$ cat example.sv 
module example(input clk);
parameter int Y = `ABS_MACRO(-5); // macro line
always_ff @(posedge clk) q = 2;
assign m = 2;
endmodule
$ verible-verilog-lint macros.sv example.sv 
example.sv:3:26: Use blocking assignments, at most, for locals inside 'always_ff' sequential blocks. [Style: sequential-logic] [always-ff-non-blocking]
```
So far so good. 
What if I introduce the same syntax error as before: 
```
$ cat example.sv 
module example(input clk);
parameter int Y = `ABS_MACRO(-5); // macro line
always_ff @(posedge clk) q = 2;
assign m  2;
endmodule
samimia@SCLSAMIMIADT01:~/git/hsp_root/dr/design/hspcommon/macros/rtl ‹dr-1› (python-3.9.6)
$ verible-verilog-lint macros.sv example.sv 
example.sv:2:19-28: preprocessing error at token "`ABS_MACRO" : Error expanding macro identifier, might not be defined before.
```
Here I would expect to the error `example.sv:5:11: syntax error at token "2"`, not `example.sv:2:19-28: preprocessing error at token "`ABS_MACRO"`

IEncinas10, 2024-05-11:
Thanks for the detailed explanation. I had hoped it would be a simple configuration error in the LSP.

verible doesn't work all that well with macros. I think it was originally intended to handle already preprocessed code.

I'll have a look just in case this can be easily fixed, but I'm not sure it will. Thanks for the report :)