chipsalliance/verible #2182: Verible formatter fails to handle comma-at-beginning-of-line style and/or inline comments

paul-demo, 2024-05-16:
v0.0-3454-g7fdc2876
Commit	2023-12-20
Built	2023-12-20T15:18:36Z

**Test case**

```systemverilog
module my_module (
        input   logic                           clk_pl_100 // a comment
    , input   logic                          aresetn // another comment

    , output a
    , output b
    );
endmodule
```

Include any options or configuration used.

**Actual output**

```
Already formatted, no change.
```

OR

```
Re-formatted text does not match formatted text; formatting failed to converge!  Please file a bug.
```

This fails to format the file unless any of the following is done:
1) I remove an underscore from `clk_pl_100`
2) I remove the blank line in the input list
3) I remove the inline comment `// another comment`

It's very unusual -- something is clearly tripping up verible in this simplified example.

paul-demo, 2024-05-16:
Trying this out with some more examples, it seems to be related to the blank line after the first two arguments, and not related to the comments:

Something like this seems to fail to format:
```systemverilog
module my_module (
        input   logic                           a
    , input   logic                          b

    , output c
    , output d
    );
endmodule
```

While putting the blank line either between a and b or between c and d both work:
```systemverilog
module my_module (
        input   logic                           a

    , input   logic                          b
    , output c
    , output d
    );
endmodule
```

```systemverilog
module my_module (
        input   logic                           a
    , input   logic                          b
    , output c

    , output d
    );
endmodule
```

kbrunham-intel, 2026-09-22:
@hzeller I can take this issue.