chipsalliance/verible #2217: It Can NOT preprocess properly in SystemVerilog files?

zpxy, 2024-07-17:
**Describe the bug**
I use a gloabal _"define"_  file   and "macro" file to manager all parameters and macros, but verible-verilog_lint  can't recongnize the macro in other file , it say an error: **error in expanding the macor , might not be difined before**
[image]

so would you plan to solve the preprocess problem?

hzeller, 2024-07-17:
Yes, preprocessor does need some work. It is fairly high on my list, but unfortunately there are other projects also competing for my attention.

MinhTran0911, 2024-12-19:
I also face the same issue, especially when working with UVM since UVM relies heavily on macros. Is there a rough estimation on when this will be fixed?