chipsalliance/verible #2221: Why is `$random` considered a "forbidden system function or task"?

benjamin051000, 2024-08-02:
`invalid-system-task-function` states:
> Checks that no forbidden system tasks or functions are used. These consist of the following functions: `$psprintf`, `$random`, and `$dist_*`. As well as non-LRM function `$srandom`. See [Style: forbidden-system-functions].

When I use `$random` in a SystemVerilog file, Verible prints this warning:
```
1. $random is a forbidden system function or task, please use $urandom instead [Style: forbidden-system-functions][invalid-system-task-function]
```

I'm curious as to why it's discouraged to use `$random` and instead use `$urandom`. Forgive me, I'm pretty new to verilog & systemverilog. Thanks for the clarification!

hzeller, 2024-08-05:
I think that was mostly a request by DV engineers that like to inject better ways to get test vector, though it would probably good to have that as an explicit style guide somewhere so that it is easy to refer to.

Maybe @fangism remembers the details ?

snsokolov, 2024-08-07:
$random breaks reproducibility:

Quote from the style guide: 
"$random and $dist_*: these functions should not be used as they are not part of the SystemVerilog random stability model and may break simulation reproducibility. Use $urandom or randomize() instead."