chipsalliance/verible #2238: It is not possible to format the code shown below

17Reset, 2024-08-19:
[pcie5_cfgchk.zip](https://github.com/user-attachments/files/16656314/pcie5_cfgchk.zip)

matlupi, 2024-08-27:
@17Reset please provide a minimal example in the issue, following the provided template.
A zip archive is not the best in terms of security, so I doubt you'll find someone opening it.

17Reset, 2024-08-29:
```verilog
`timescale 1 ns / 1 ps
module
pcie5_eqif

(

input
wire
pclk
,

input
wire
prstn
,

input
wire
psrst
,

input
wire
[`PCIE_TESTIN_SIZE-1:0]
test_in
,

input
wire
equ_reset
,

input
wire
chgspeed_done
,

input
wire
[`EQRESP_SIZE-1:0]
eq_resp
,

output
wire
[`EQCMD_SIZE-1:0]
eq_cmd
,

input
wire
req_start
,

input
wire
[2:0]
req_cmd
,

output
reg
[2:0]
req_resp
,

output
wire
req_done
[... 1458 lines cut ...]
+:
3]

==

`RESP_EQUFAIL

)

?
1'b1
:
1'b0
;

assign
test_out_eqif
[127]
=
1'b0
;

endmodule

```

17Reset, 2024-08-29:
Run the above code with the command `verible-verilog-format.exe demo1.v` and the console output，however the above is the result of an obfuscate from another programme:
```
demo1.v: Formatted output is lexically different from the input.    Please file a bug.  Details:
Mismatched token enums.  got: (MacroIdItem) (#298: "`RESP_IDLE") vs. (MacroIdentifier) (#296: "`RESP_IDLE")
First mismatched token [295]: (MacroIdItem) (#298: "`RESP_IDLE") vs. (MacroIdentifier) (#296: "`RESP_IDLE")
; problematic formatter output is
```

17Reset, 2024-09-02:
There is also the problem shown below, which is the result of a code sample and format:
```
assign test_out_eqif[113] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_SETINITPRESET) ? 1'b1 : 1'b0;
assign test_out_eqif [114] =  ( eq_cmd [`EQCMD_COMMAND +: 3]  ==  `CMD_REQLOCALPRESET  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif[115] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_REQLOCALCOEFF) ? 1'b1 : 1'b0;
assign test_out_eqif[116] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_REMOTECOEFF) ? 1'b1 : 1'b0;
assign test_out_eqif[117] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_PERFORMRXEVAL) ? 1'b1 : 1'b0;
assign test_out_eqif [118] =  ( eq_cmd [`EQCMD_COMMAND +: 3]  ==  `CMD_REMOTEREQREJECT  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif[119] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_REMOTEREQFAIL) ? 1'b1 : 1'b0;
assign test_out_eqif [120] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_LOCALCOEFFMAPPING  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif [121] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_LOCALCOEFFACCEPT  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif [122] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_LOCALCOEFFREJECT  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif [123] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_NEWREMOTEPRESET  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif [124] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_NEWREMOTECOEFF  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif [125] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_EQUSUCCESS  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif[126] = (eq_resp[`EQRESP_RESPONSE+:3] == `RESP_EQUFAIL) ? 1'b1 : 1'b0;
assign test_out_eqif[127] = 1'b0;
    ```
    
    ```
assign test_out_eqif[113] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_SETINITPRESET) ? 1'b1 : 1'b0;
assign test_out_eqif[114] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_REQLOCALPRESET) ? 1'b1 : 1'b0;
assign test_out_eqif[115] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_REQLOCALCOEFF) ? 1'b1 : 1'b0;
assign test_out_eqif[116] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_REMOTECOEFF) ? 1'b1 : 1'b0;
assign test_out_eqif[117] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_PERFORMRXEVAL) ? 1'b1 : 1'b0;
assign test_out_eqif[118] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_REMOTEREQREJECT) ? 1'b1 : 1'b0;
assign test_out_eqif[119] = (eq_cmd[`EQCMD_COMMAND+:3] == `CMD_REMOTEREQFAIL) ? 1'b1 : 1'b0;
assign test_out_eqif [120] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_LOCALCOEFFMAPPING  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif [121] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_LOCALCOEFFACCEPT  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif [122] =  ( eq_resp [`EQRESP_RESPONSE +: 3]  ==  `RESP_LOCALCOEFFREJECT  )  ? 1'b1 : 1'b0 ;
assign test_out_eqif[123] = (eq_resp[`EQRESP_RESPONSE+:3] == `RESP_NEWREMOTEPRESET) ? 1'b1 : 1'b0;
assign test_out_eqif[124] = (eq_resp[`EQRESP_RESPONSE+:3] == `RESP_NEWREMOTECOEFF) ? 1'b1 : 1'b0;
assign test_out_eqif[125] = (eq_resp[`EQRESP_RESPONSE+:3] == `RESP_EQUSUCCESS) ? 1'b1 : 1'b0;
assign test_out_eqif[126] = (eq_resp[`EQRESP_RESPONSE+:3] == `RESP_EQUFAIL) ? 1'b1 : 1'b0;
assign test_out_eqif[127] = 1'b0;
    ```