chipsalliance/verible #2248: Lint rule [instance-shadowing] raising violations on modport entries

sconwayaus, 2024-09-14:
**Describe the bug**
Lint rule 'instance-shadowing' is not handling interfaces well, rasising vilations on modport items shadowing interface signals.

**Test case (preferably reduced)**
```systemverilog
// Ran the style linter on this code.
interface rconst_interface;
    logic [23:0] i;
    logic [63:0] rc;

    modport slave (
        input i,    // Violation raised here for 'i'
        output rc    // Violation raised here for 'rc'
    );
endinterface

```
**Actual vs. expected behavior**
False positive violations are raised
Symbol `i` is shadowing symbol `i` defined at @ [Style: mark-shadowed-instances][instance-shadowing]
Symbol `rc` is shadowing symbol `rc` defined at @ [Style: mark-shadowed-instances][instance-shadowing]

IEncinas10, 2024-09-21:
I took a look at this. I guess the best solution is to ignore declarations inside modports given that they are not actually shadowing anything, right? 

Let me know if you're planning on opening a PR

sconwayaus, 2024-09-21:
Thanks. I took a look at this a few days ago and like you said, it was just
a matter of  ignoring symbols in modports. I'll push something across
shortly.

PR #2255 (2024-09-23): Instance shadowing fix modport
This PR resolves #2248, fixing the issue with the lint rule 'instance-shadowing' raising violations on modport items.

The change simply ignores symbols in modports, both simple and clocking ports.
closed 2024-09-23 by fangism