chipsalliance/verible #2253: Getting Error Unsupported: timing control statement in this location, for wait and @(posedge) statements

v4ibhav, 2024-09-19:
IDE Version: NVIM v0.10.1
Tools: Neovim with Verible LSP
Additional Plugins: Veridian

**Failed**
Linting failed with Verible LSP while handling timing control statements like @(posedge clk) and wait.
**Expectation**
Verible LSP should handle timing control and wait statements without errors in Neovim.

I am getting unsupported timing control statement in this location and Unsupported wait statements. Verible LSP flagged valid SystemVerilog code with errors, specifically regarding timing control constructs.

[image]

sconwayaus, 2024-09-21:
The text in RED, is that the error message?

I tried a couple of lines in the latest verible language server and I don't get these issues. I'm suspicious that you issue is something to do with Veridian.

What do you get if you run `verible-verilog-lint --ruleset=all some_file.sv`?

v4ibhav, 2024-09-21:
This is what I get when I run the command. Yes, you are correct, the issue is not with verible but with veridian. Once I removed veridian, I no longer encountered those errors.

conv_testbench.sv:2:8-13: Declared module does not match the first dot-delimited component of file name: "conv_testbench" [Style: file-names] [module-filename]
conv_testbench.sv:5:3-11: 'parameter' declarations should only be in the formal parameter list of modules/classes. [Style: constants] [proper-parameter-declaration]
conv_testbench.sv:5:13-21: Explicitly define a storage type for every parameter and localparam, (CLK_PHASE). [Style: constants] [explicit-parameter-storage-type]
conv_testbench.sv:6:3-11: 'parameter' declarations should only be in the formal parameter list of modules/classes. [Style: constants] [proper-parameter-declaration]
conv_testbench.sv:6:13-17: Explicitly define a storage type for every parameter and localparam, (ROUND). [Style: constants] [explicit-parameter-storage-type]
conv_testbench.sv:8:16-27: Signal name does not match the naming convention defined by regex pattern: [a-z_0-9]+ [Style: signal-conventions] [signal-name-style]
conv_testbench.sv:9:16-27: Signal name does not match the naming convention defined by regex pattern: [a-z_0-9]+ [Style: signal-conventions] [signal-name-style]
conv_testbench.sv:10:16-25: Signal name does not match the naming convention defined by regex pattern: [a-z_0-9]+ [Style: signal-conventions] [signal-name-style]
conv_testbench.sv:11:16-26: Signal name does not match the naming convention defined by regex pattern: [a-z_0-9]+ [Style: signal-conventions] [signal-name-style]
conv_testbench.sv:12:16-24: Signal name does not match the naming convention defined by regex pattern: [a-z_0-9]+ [Style: signal-conventions] [signal-name-style]
conv_testbench.sv:13:16-22: Signal name does not match the naming convention defined by regex pattern: [a-z_0-9]+ [Style: signal-conventions] [signal-name-style]
conv_testbench.sv:15:16-28: Signal name does not match the naming convention defined by regex pattern: [a-z_0-9]+ [Style: signal-conventions] [signal-name-style]
conv_testbench.sv:92:5-11: forever block constructs shall explicitly use begin/end. Expected begin, got # [Style: explicit-begin] [explicit-begin]
conv_testbench.sv:92:14-22: Symbol `CLK_PHASE` is shadowing symbol `CLK_PHASE` defined at conv_testbench.sv:5:13 [Style: mark-shadowed-instances] [instance-shadowing]

sconwayaus, 2024-09-22:
Good to hear you have it sorted.