chipsalliance/verible #2278: Improve lint rule help

IEncinas10, 2024-10-06:
**Summary**

Rule descriptions can be obtained by running `verible-verilog-lint --help_rules=all`, which is 
not properly formatted for long descriptions. It also doesn't support showing examples, which might be useful for some complex rules. Current workaround is reading the source code and tests for a rule, which is not all that great.

**Proposal**

1. Fixing formatting for very long descriptions shouldn't be hard. Easiest solution is to break description strings with newlines and indentation after a certain threshold is past.
2. We might want to add an extra-verbose mode where rule help prints a few examples for a given rule.

batmnnn, 2024-11-09:
can i  assign this issue to me ? i'm able to do the first part of proposal , will need some help in second @IEncinas10

IEncinas10, 2024-11-09:
Sure! They don't have to be completed in the same PR, just part 1) would be great.

Bhavin-umatiya, 2024-12-24:
@IEncinas10 Hello
please check any further change so u can guide me i'll try