chipsalliance/verible #2288: how to change lint rule "legacy-generate-region"?

ddppt-yy, 2024-11-07:
**Summary**
//Brief description of request. Reference `[style-lint-rule-name]` if this is
//about an existing rule, or suggest new lint rule name.

in rule file https://chipsalliance.github.io/verible/lint.html#generate-label-prefix , there is no parameter to user change the  `style_regex`.

May I ask whether modification is not supported or I have not found a way to modify

IEncinas10, 2024-11-07:
The rule specifically checks against `gen_` or `g_` prefixes https://github.com/chipsalliance/verible/blob/86ee9bab719daed109cf399454f1b486b03900b5/verilog/analysis/checkers/generate_label_prefix_rule.cc#L86

The rule could be generalized to take a regex (or a list of prefixes, ...). In case you're considering giving it a try, this might serve as an example: https://github.com/chipsalliance/verible/blob/master/verilog/analysis/checkers/enum_name_style_rule.cc

ddppt-yy, 2024-11-08:
> The rule specifically checks against `gen_` or `g_` prefixes [...]

This means that the user cannot modify the regular expression with test parameters, only by recompiling it to a new linter file, right?

IEncinas10, 2024-11-08:
Right. Currently, there is no way of configuring the rule. But it can be **made** configurable.

hzeller, 2024-11-11:
I think we should keep this open until implemented.

pranjal3060, 2024-12-05:
May I work on this issue

closed 2026-09-10 by hzeller