chipsalliance/verible #2289: Formatter is confused by macros

isaacde, 2024-11-07:
**Test case**

I'm trying to get the formatter working with legacy verilog code. Here is a simple example that will generate the syntax parsing / token issues I've seen:

```systemverilog
`define FDLY 0.1
`define always_pe_ne(clk,rst_n) always @(posedge clk or negedge rst_n)

module verible_formatter_token_defines #(
) (
    input logic clk,
    input logic rst_n
);

  logic [3:0] countA, countB, countC, countD;

  always_ff @(posedge clk, negedge rst_n) begin
    if (!rst_n) begin
    countA <= 0;
    countB <= 0;
    end else begin
      countA <= #(`FDLY) (countA + 1);
      countB <= #`FDLY (countB + 1); // verible_formatter_token_defines.sv:18:18-22: syntax error at token "`FDLY"
    end
  end

  `always_pe_ne(clk, rst_n) 
  begin
    if (~rst_n) begin
      countC <= 0;
    end else begin
      countC <= countC + 1;
    end
  end

  `always_pe_ne(clk, rst_n) begin
    if (~rst_n) begin
      countD <= 0;
    end else begin
      countD <= countD + 1;
    end
  end
endmodule

```

`verible-verilog-format` diagnostic output:

```systemverilog
$ verible-verilog-format verible_formatter_token_defines.sv
I1107 08:40:21.374077   20811 verilog_analyzer.cc:132] AnalyzeAutomaticMode
I1107 08:40:21.374202   20811 text_structure.cc:393] FastLineRangeConsistencyCheck
I1107 08:40:21.374273   20811 text_structure.cc:338] FastTokenRangeConsistencyCheck
I1107 08:40:21.374293   20811 text_structure.cc:408] SyntaxTreeConsistencyCheck
I1107 08:40:21.374311   20811 text_structure.cc:393] FastLineRangeConsistencyCheck
I1107 08:40:21.374331   20811 text_structure.cc:338] FastTokenRangeConsistencyCheck
I1107 08:40:21.374351   20811 text_structure.cc:408] SyntaxTreeConsistencyCheck
I1107 08:40:21.375039   20811 parser_param.cc:51] verible_formatter_token_defines.sv: recovered syntax error: (#297: "`FDLY")
I1107 08:40:21.375091   20811 bison_parser_common.cc:41] verible_formatter_token_defines.sv: verilog_parse error: syntax error
I1107 08:40:21.375158   20811 parser_param.cc:51] verible_formatter_token_defines.sv: recovered syntax error: (#663: "<=")
I1107 08:40:21.375192   20811 bison_parser_common.cc:41] verible_formatter_token_defines.sv: verilog_parse error: syntax error
I1107 08:40:21.375221   20811 parser_param.cc:51] verible_formatter_token_defines.sv: recovered syntax error: (#663: "<=")
I1107 08:40:21.375284   20811 bison_parser_common.cc:41] verible_formatter_token_defines.sv: verilog_parse error: syntax error
I1107 08:40:21.375318   20811 parser_param.cc:51] verible_formatter_token_defines.sv: recovered syntax error: (#320: "begin")
I1107 08:40:21.375367   20811 bison_parser_common.cc:41] verible_formatter_token_defines.sv: verilog_parse error: syntax error
I1107 08:40:21.375391   20811 parser_param.cc:51] verible_formatter_token_defines.sv: recovered syntax error: (#334: "end")
I1107 08:40:21.375441   20811 bison_parser_common.cc:41] verible_formatter_token_defines.sv: verilog_parse error: syntax error
I1107 08:40:21.375487   20811 parser_param.cc:51] verible_formatter_token_defines.sv: recovered syntax error: (#334: "end")
I1107 08:40:21.375523   20811 bison_parser_common.cc:41] verible_formatter_token_defines.sv: verilog_parse error: syntax error
I1107 08:40:21.375565   20811 bison_parser_adapter.h:53] max_used_stack_size : 0
I1107 08:40:21.375603   20811 verilog_analyzer.cc:157] Error analyzing verilog.
I1107 08:40:21.375625   20811 verilog_analyzer.cc:167] Retrying parsing in mode: "".
I1107 08:40:21.375646   20811 verilog_analyzer.cc:200] end of AnalyzeAutomaticMode
I1107 08:40:21.375700   20811 text_structure.cc:393] FastLineRangeConsistencyCheck
I1107 08:40:21.375733   20811 text_structure.cc:338] FastTokenRangeConsistencyCheck
I1107 08:40:21.375757   20811 text_structure.cc:408] SyntaxTreeConsistencyCheck
`define FDLY 0.1
`define always_pe_ne(clk,rst_n) always @(posedge clk or negedge rst_n)

module verible_formatter_token_defines #(
) (
    input logic clk,
    input logic rst_n
);

  logic [3:0] countA, countB, countC, countD;

  always_ff @(posedge clk, negedge rst_n) begin
    if (!rst_n) begin
    countA <= 0;
    countB <= 0;
    end else begin
      countA <= #(`FDLY) (countA + 1);
      countB <= #`FDLY (countB + 1);
    end
  end

  `always_pe_ne(clk, rst_n) 
  begin
    if (~rst_n) begin
      countC <= 0;
    end else begin
      countC <= countC + 1;
    end
  end

  `always_pe_ne(clk, rst_n) begin
    if (~rst_n) begin
      countD <= 0;
    end else begin
      countD <= countD + 1;
    end
  end
endmodule
verible_formatter_token_defines.sv: verible_formatter_token_defines.sv:18:18-22: syntax error at token "`FDLY"
verible_formatter_token_defines.sv:25:14-15: syntax error at token "<="
verible_formatter_token_defines.sv:27:14-15: syntax error at token "<="
verible_formatter_token_defines.sv:31:29-33: syntax error at token "begin"
verible_formatter_token_defines.sv:34:5-7: syntax error at token "end"
verible_formatter_token_defines.sv:36:5-7: syntax error at token "end"
```

**Expected behavior**

- brackets around the RHS after ` <= #delay` should not behave differently than if there were no brackets
- `begin` should be paired with `end` and whether it is on the same line or next line to a macro should not cause syntax errors

Brughy, 2025-03-04:
Probably it is related to same question, but i found this issue : 

Given the followign moduel a.sv :

`define __AXILITE_DECL( PREFIX, AW, DW, RQ_QUAL, RP_QUAL, TERM ) \
   RQ_QUAL [AW-1:0]     PREFIX``_awaddr  TERM \
   RQ_QUAL              PREFIX``_awvalid  TERM \
   RP_QUAL              PREFIX``_awready  TERM \
   RQ_QUAL [2:0]        PREFIX``_awprot  TERM \
   RQ_QUAL [DW-1:0]     PREFIX``_wdata  TERM \
   RQ_QUAL [DW/8-1:0]   PREFIX``_wstrb  TERM \
   RQ_QUAL              PREFIX``_wvalid  TERM \
   RP_QUAL              PREFIX``_wready  TERM \
   RP_QUAL [1:0]        PREFIX``_bresp  TERM \
   RQ_QUAL              PREFIX``_bready  TERM \
   RP_QUAL              PREFIX``_bvalid  TERM \
   RQ_QUAL [AW-1:0]     PREFIX``_araddr  TERM \
   RQ_QUAL [2:0]        PREFIX``_arprot  TERM \
   RQ_QUAL              PREFIX``_arvalid  TERM \
   RP_QUAL              PREFIX``_arready  TERM \
   RQ_QUAL              PREFIX``_rready  TERM \
   RP_QUAL [DW-1:0]   PREFIX``_rdata  TERM \
   RP_QUAL [1:0]        PREFIX``_rresp  TERM \
   RP_QUAL              PREFIX``_rvalid  TERM

module a
(
    input wire i_clk
);

     `__AXILITE_DECL( s, 32, 32, wire, wire, ; )

endmodule

Launching...
**verible-verilog-preprocessor preprocess a.sv**

The output is:

 `define__AXILITE_DECL(PREFIX,AW,DW,RQ_QUAL,RP_QUAL,TERM)\
   RQ_QUAL [AW-1:0]     PREFIX``_awaddr  TERM \
   RQ_QUAL              PREFIX``_awvalid  TERM \
   RP_QUAL              PREFIX``_awready  TERM \
   RQ_QUAL [2:0]        PREFIX``_awprot  TERM \
   RQ_QUAL [DW-1:0]     PREFIX``_wdata  TERM \
   RQ_QUAL [DW/8-1:0]   PREFIX``_wstrb  TERM \
   RQ_QUAL              PREFIX``_wvalid  TERM \
   RP_QUAL              PREFIX``_wready  TERM \
   RP_QUAL [1:0]        PREFIX``_bresp  TERM \
   RQ_QUAL              PREFIX``_bready  TERM \
   RP_QUAL              PREFIX``_bvalid  TERM \
   RQ_QUAL [AW-1:0]     PREFIX``_araddr  TERM \
   RQ_QUAL [2:0]        PREFIX``_arprot  TERM \
   RQ_QUAL              PREFIX``_arvalid  TERM \
   RP_QUAL              PREFIX``_arready  TERM \
   RQ_QUAL              PREFIX``_rready  TERM \
   RP_QUAL [DW-1:0]   PREFIX``_rdata  TERM \
   RP_QUAL [1:0]        PREFIX``_rresp  TERM \
   RP_QUAL              PREFIX``_rvalid  TERM

module a
(
    input wire i_clk
);

     \
wire[32-1:0]s``_awaddr;\
wires``_awvalid;\
wires``_awready;\
wire[2:0]s``_awprot;\
wire[32-1:0]s``_wdata;\
wire[32/8-1:0]s``_wstrb;\
wires``_wvalid;\
wires``_wready;\
wire[1:0]s``_bresp;\
wires``_bready;\
wires``_bvalid;\
wire[32-1:0]s``_araddr;\
wire[2:0]s``_arprot;\
wires``_arvalid;\
wires``_arready;\
wires``_rready;\
wire[32-1:0]s``_rdata;\
wire[1:0]s``_rresp;\
wires``_rvalid;

endmodule

The output it is not correct

Brughy, 2025-03-04:
Followiong the previus arguments i report second example to analyze in deep this question.

Given the following module aa.sv :

```
 `define __AXILITE_DECL( PREFIX, AW, DW, RQ_QUAL, RP_QUAL, TERM ) \
   RQ_QUAL [AW-1:0]     PREFIX``_awaddr  TERM \
   RQ_QUAL              PREFIX``_awvalid  TERM \
   RP_QUAL              PREFIX``_awready  TERM \
   RQ_QUAL [2:0]        PREFIX``_awprot  TERM \
   RQ_QUAL [DW-1:0]     PREFIX``_wdata  TERM \
   RQ_QUAL [DW/8-1:0]   PREFIX``_wstrb  TERM \
   RQ_QUAL              PREFIX``_wvalid  TERM \
   RP_QUAL              PREFIX``_wready  TERM \
   RP_QUAL [1:0]        PREFIX``_bresp  TERM \
   RQ_QUAL              PREFIX``_bready  TERM \
   RP_QUAL              PREFIX``_bvalid  TERM \
   RQ_QUAL [AW-1:0]     PREFIX``_araddr  TERM \
   RQ_QUAL [2:0]        PREFIX``_arprot  TERM \
   RQ_QUAL              PREFIX``_arvalid  TERM \
   RP_QUAL              PREFIX``_arready  TERM \
   RQ_QUAL              PREFIX``_rready  TERM \
   RP_QUAL [DW-1:0]     PREFIX``_rdata  TERM \
   RP_QUAL [1:0]        PREFIX``_rresp  TERM \
   RP_QUAL              PREFIX``_rvalid  TERM

 `define __AXILITE_PORT_MAP_DECL( PORT_PREFIX, SIG_PREFIX ) \
    .PORT_PREFIX``_awaddr  ( SIG_PREFIX``_awaddr),  \
    .PORT_PREFIX``_awvalid ( SIG_PREFIX``_awvalid), \
    .PORT_PREFIX``_awready ( SIG_PREFIX``_awready), \
    .PORT_PREFIX``_awprot  ( SIG_PREFIX``_awprot),  \
    .PORT_PREFIX``_wdata   ( SIG_PREFIX``_wdata),   \
    .PORT_PREFIX``_wvalid  ( SIG_PREFIX``_wvalid),  \
    .PORT_PREFIX``_wready  ( SIG_PREFIX``_wready),  \
    .PORT_PREFIX``_wstrb   ( SIG_PREFIX``_wstrb),   \
    .PORT_PREFIX``_bready  ( SIG_PREFIX``_bready),  \
    .PORT_PREFIX``_bvalid  ( SIG_PREFIX``_bvalid),  \
    .PORT_PREFIX``_bresp   ( SIG_PREFIX``_bresp),   \
    .PORT_PREFIX``_araddr  ( SIG_PREFIX``_araddr),  \
    .PORT_PREFIX``_arvalid ( SIG_PREFIX``_arvalid), \
    .PORT_PREFIX``_arready ( SIG_PREFIX``_arready), \
    .PORT_PREFIX``_arprot  ( SIG_PREFIX``_arprot),  \
    .PORT_PREFIX``_rready  ( SIG_PREFIX``_rready),  \
    .PORT_PREFIX``_rvalid  ( SIG_PREFIX``_rvalid),  \
    .PORT_PREFIX``_rdata   ( SIG_PREFIX``_rdata),   \
    .PORT_PREFIX``_rresp   ( SIG_PREFIX``_rresp),

module aa
(
    output o_port1
    ,input wire i_clk1
    ,input wire i_rst1
);

     `__AXILITE_DECL( s, 32, 32, wire, wire, ; )

    a
    a_i (
        `__AXILITE_PORT_MAP_DECL (i_x, s)
        .o_port ( o_port1 ),
        .i_clk ( i_clk1 )
    );

endmodule
```

Launching...
verible-verilog-preprocessor preprocess as.sv

The output is:
[... 102 lines cut ...]

The errored output is:

```
aa.sv:1:67: syntax error at token "\"
aa.sv:2:47: syntax error at token "\"
aa.sv:3:48: syntax error at token "\"
aa.sv:4:48: syntax error at token "\"
aa.sv:5:47: syntax error at token "\"
aa.sv:6:46: syntax error at token "\"
aa.sv:7:46: syntax error at token "\"
aa.sv:8:47: syntax error at token "\"
aa.sv:9:47: syntax error at token "\"
aa.sv:10:46: syntax error at token "\"
aa.sv:11:47: syntax error at token "\"
aa.sv:12:47: syntax error at token "\"
aa.sv:13:47: syntax error at token "\"
aa.sv:14:47: syntax error at token "\"
aa.sv:15:48: syntax error at token "\"
aa.sv:16:48: syntax error at token "\"
aa.sv:17:47: syntax error at token "\"
aa.sv:18:46: syntax error at token "\"
aa.sv:19:46: syntax error at token "\"
aa.sv:22:61: syntax error at token "\"
```

Brughy, 2025-03-07:
Some error also are propagated into analysis7checkers/module-instantiation-rules.cc with uncorrect number port count