chipsalliance/verible #2290: Allow verible.filelist per GitHub submodule

thomas-woehrle, 2024-11-11:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP: VScode
 - IDE version: 1.95.2
 - What other SystemVerilog plugins are active alongside: Verilog Hightlight. Source: https://github.com/tzyLee/verilog-highlight
-----

**What activity failed**
_(what functionality did not work as expected (e.g. Linting, Formatting, AUTO-expansion, go-to-definition, outline, hover,...)_
go-to-definition
**Expectation**
I have a repository with git submodules. Each submodule has its own verible.filelist file. When working in my repository, I would want to go-to-definition on the basis of the verible.filelist of the git submodule which a file is part of. 
**What actually happened**
No go-to-definition via cmd-click is possible. Presumably because verible language server only takes into account the verible.filelist file in the root of the project, but not the verible.filelist files in the roots of the git submodules.

IEncinas10, 2024-12-27:
Hello! What do you think of coming up with a prefix to build a filelist from others?

Imagine you have two filelists: `verible.filelist` and `my_submodule/verible.filelist`. Then, you would link them by adding the following line into `verible.filelist` (the top one)

```
+filelist+./my_submodule/verible.filelist
```

glaserf, 2025-02-03:
Is it really so hard to generate those file lists automatically (from verible itself)? Having to maintain them is something I'm not used to from other language servers (e.g., the one of the verilog extension of Sublime Text) and also quite cumbersome in projects that frequently change.