chipsalliance/verible #2313: language server crashes periodically

gobbedy, 2024-12-25:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP ?
 -Vscode
 - IDE version:
 - 1.96.2
 - What other SystemVerilog plugins are active alongside ?
 - "Verilog-HDL/SystemVerilog/Bluespec SystemVerilog"
-----

**What activity failed**
_(what functionality did not work as expected (e.g. Linting, Formatting, AUTO-expansion, go-to-definition, outline, hover,...)_

I have no idea what the feature is, but the language server crashes out periodically, and the Output window grabs focus, which kills my workflow. I'm sure I'm doing something wrong, but the error message is difficult for me to make sense of.

Can you please help?

Here is the content of the Output window when it fails

Verible Verilog Language Server built at v0.0-3880-g91a8757d
commandline: /home/gperraul/.local/bin/verible-verilog-ls --rules_config /home/gperraul/.verible/rules.lint 
F1224 19:48:15.895341 2412459 symbol-table.cc:1524] Check failed: ABSL_DIE_IF_NULL(ref.components)->Children().size() == 1 (0 vs. 1) 
*** Check failure stack trace: ***
    @           0x5e9a2d  absl::lts_20240116::log_internal::LogMessageFatal::~LogMessageFatal()
    @           0x457020  verilog::SymbolTable::Builder::LookupOrInjectOutOfLineDefinition()
    @           0x45713c  verilog::SymbolTable::Builder::HandleQualifiedId()
    @           0x45a6df  verilog::SymbolTable::Builder::Visit()
    @           0x5db661  verible::TreeContextVisitor::Visit()
    @           0x45a3cf  verilog::SymbolTable::Builder::Visit()
    @           0x5db661  verible::TreeContextVisitor::Visit()
    @           0x45a34f  verilog::SymbolTable::Builder::Visit()
    @           0x5db661  verible::TreeContextVisitor::Visit()
    @           0x45a33b  verilog::SymbolTable::Builder::Visit()
    @           0x45523e  verilog::BuildSymbolTable()
    @           0x45532d  verilog::ParseFileAndBuildSymbolTable()
    @           0x455663  verilog::SymbolTable::Build()
    @           0x44a511  verilog::SymbolTableHandler::BuildProjectSymbolTable()
    @           0x44a664  verilog::SymbolTableHandler::Prepare()
    @           0x44a6fb  verilog::SymbolTableHandler::FindDefinitionLocation()
    @           0x4256c2  std::_Function_handler<>::_M_invoke()
    @           0x584c80  verible::lsp::JsonRpcDispatcher::CallRequestHandler()
    @           0x58553d  verible::lsp::JsonRpcDispatcher::DispatchMessage()
    @           0x585963  verible::lsp::MessageStreamSplitter::ProcessContainedMessages()
    @           0x585ed9  verible::lsp::MessageStreamSplitter::ReadInput()
    @           0x585fad  verible::lsp::MessageStreamSplitter::PullFrom()
    @           0x41fedf  verilog::VerilogLanguageServer::Run()
    @           0x416712  main
    @           0x690f3a  __libc_start_call_main
*** SIGABRT received at time=1735087695 on cpu 6 ***
PC: @           0x6c8c2c  (unknown)  __pthread_kill
    @           0x6a8de0  (unknown)  (unknown)
[Info  - 7:48:16 PM] Connection to server got closed. Server will restart.
[Error - 7:48:16 PM] Request textDocument/definition failed.
Error: Connection got disposed.
	at Object.dispose (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:195828)
	at Object.dispose (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:275922)
	at _.handleConnectionClosed (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:276135)
	at _.handleConnectionClosed (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:336506)
	at t (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:274224)
	at i.invoke (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:197442)
	at o.fire (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:198203)
	at Y (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:185086)
	at i.invoke (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:197442)
	at o.fire (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:198203)
	at m.fireClose (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:206085)
	at Socket.<anonymous> (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:207670)
	at Socket.emit (node:events:530:35)
	at Pipe.<anonymous> (node:net:343:12)
Verible Verilog Language Server built at v0.0-3880-g91a8757d
commandline: /home/gperraul/.local/bin/verible-verilog-ls --rules_config /home/gperraul/.verible/rules.lint 
F1224 19:50:36.951039 2412596 symbol-table.cc:1524] Check failed: ABSL_DIE_IF_NULL(ref.components)->Children().size() == 1 (0 vs. 1) 
*** Check failure stack trace: ***
    @           0x5e9a2d  absl::lts_20240116::log_internal::LogMessageFatal::~LogMessageFatal()
    @           0x457020  verilog::SymbolTable::Builder::LookupOrInjectOutOfLineDefinition()
    @           0x45713c  verilog::SymbolTable::Builder::HandleQualifiedId()
    @           0x45a6df  verilog::SymbolTable::Builder::Visit()
    @           0x5db661  verible::TreeContextVisitor::Visit()
    @           0x45a3cf  verilog::SymbolTable::Builder::Visit()
    @           0x5db661  verible::TreeContextVisitor::Visit()
    @           0x45a34f  verilog::SymbolTable::Builder::Visit()
    @           0x5db661  verible::TreeContextVisitor::Visit()
    @           0x45a33b  verilog::SymbolTable::Builder::Visit()
    @           0x45523e  verilog::BuildSymbolTable()
    @           0x45532d  verilog::ParseFileAndBuildSymbolTable()
    @           0x455663  verilog::SymbolTable::Build()
    @           0x44a511  verilog::SymbolTableHandler::BuildProjectSymbolTable()
    @           0x44a664  verilog::SymbolTableHandler::Prepare()
    @           0x44a6fb  verilog::SymbolTableHandler::FindDefinitionLocation()
    @           0x4256c2  std::_Function_handler<>::_M_invoke()
    @           0x584c80  verible::lsp::JsonRpcDispatcher::CallRequestHandler()
    @           0x58553d  verible::lsp::JsonRpcDispatcher::DispatchMessage()
    @           0x585963  verible::lsp::MessageStreamSplitter::ProcessContainedMessages()
    @           0x585ed9  verible::lsp::MessageStreamSplitter::ReadInput()
    @           0x585fad  verible::lsp::MessageStreamSplitter::PullFrom()
    @           0x41fedf  verilog::VerilogLanguageServer::Run()
    @           0x416712  main
    @           0x690f3a  __libc_start_call_main
*** SIGABRT received at time=1735087836 on cpu 5 ***
PC: @           0x6c8c2c  (unknown)  __pthread_kill
    @           0x6a8de0  (unknown)  (unknown)
[Info  - 7:50:37 PM] Connection to server got closed. Server will restart.
[Error - 7:50:37 PM] Request textDocument/definition failed.
Error: Connection got disposed.
	at Object.dispose (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:195828)
	at Object.dispose (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:275922)
	at _.handleConnectionClosed (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:276135)
	at _.handleConnectionClosed (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:336506)
	at t (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:274224)
	at i.invoke (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:197442)
	at o.fire (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:198203)
	at Y (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:185086)
	at i.invoke (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:197442)
	at o.fire (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:198203)
	at m.fireClose (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:206085)
	at Socket.<anonymous> (/home/gperraul/.vscode-server/extensions/chipsalliance.verible-0.0.3890/dist/extension.js:2:207670)
	at Socket.emit (node:events:530:35)
	at Pipe.<anonymous> (node:net:343:12)

gobbedy, 2024-12-25:
Ok, i think i've figured out how to reproduce. This seems to happen whenever I toggle the Primary Side Bar (using Ctrl-B)

I have no idea why that would cause the verible language server to crash, but it seems to be a systematic behavior

gobbedy, 2024-12-25:
Oddly, if I toggle the Primary Side Bar using the menu (View->Appearance->Primary Side Bar), the language server does not crash. It's only when I use the Ctrl-B shortcut

gobbedy, 2024-12-25:
I just checked my keybings, and Ctrl-B appears to be only assigned to 'Toggle Primary Side Bar Visibility', so that seems to rule out the Ctrl-B shortcut having any other side effects

gobbedy, 2024-12-26:
I noticed that this happens at various other times. The one other context that I can reproduce reliably is by hitting "Ctrl-Z" in any open .sv document.

It also happens with other keyboard shortcuts but not systematically.

gobbedy, 2024-12-26:
This is a showstopper for me. Would be grateful for some help.

gobbedy, 2024-12-26:
I tried various versions of the vscode verible extensions and I get the same issue in all of them. Everything from the latest version (0.0.3890) to the oldest available version (0.0.3051)

gobbedy, 2024-12-26:
I use remote-ssh, not sure if relevant?

gobbedy, 2024-12-26:
I tried going back to the oldest available version of verible itself (ie the language server itself) using the github, ie v0.0-3795, with the corresponding 0.0.3795 version of the vscode extension. I still got the same ABSL_DIE_IF_NULL crash with a ctrl-Z

gobbedy, 2024-12-26:
I tried disabling the "Verilog-HDL/SystemVerilog/Bluespec SystemVerilog" extension and restarting my vscode, but to no avail. Still get the same verible crash

IEncinas10, 2024-12-27:
Hello! It looks like verible is having trouble building the symbol table for your code so that you can go to definitions and such. From the stack trace it seems the failure happens after you try to go to a symbol definition, could you check? 

I'll have a look just in case the problem is an easy fix. If you could share a reproducer, that would be great

gobbedy, 2024-12-28:
Hi, thanks for your reply. Like I said it happens when I do ctrl-z (and other keyboard shortcut). This will occur if I start vscode and just do ctrl-z with an open .sv file and nothing else. Note that if I don't use a keyboard shortcut to trigger the crash, it will behave properly. The linter underlines lines of code it's unhappy with and tell me what's wrong with them when I hover over them. And the formatter works properly.

As for checking what happens when I go to a symbol definition, how do I go to a symbol definition?

gobbedy, 2024-12-28:
Ok, I did right click on 'Go to definition' and it did indeed crash. Can you please help on how to fix this? Is there some logfile I can look at which will help me debug what verible struggles with?

Or can I turn off this feature? Going to definitions would be nice but I mostly care about the linter and formatter.

IEncinas10, 2024-12-28:
There must be something wrong with the tool, but it is hard to fix the crash without having the snippet of code that triggers it. If you can't provide the actual code, perhaps you can modify it enough so that you can share it?

I don't know how to disable just go-to-definition in vscode. Some editors like nvim support doing this, so perhaps there is a way. 

If you are in a hurry and disabling go-to-definition isn't possible, recompiling the language server changing this line https://github.com/chipsalliance/verible/blob/2585d6a3b181873e31d5eceaf4116cc3b7c556e4/verible/verilog/tools/ls/verilog-language-server.cc#L87 to `false` would likely fix the issue.

gobbedy, 2024-12-29:
Hi @IEncinas10 , thanks again for your helpful reply. I'm grateful for your support.

After some debugging I discovered that the code causing the crash resided in a file brought in by my verible.filelist.

My verible.filelist contains thousands of files, so there might be multiple snippets causing a crash, but I managed to isolate a minimal standalone that causes a crash:

```
class myclass;
  int a;
endclass

function void myclass::sample();
  // do something
endfunction
```

If I right click on the 'a' variable and click 'Go to definitions', the crash occurs.

Note that if I rename the function to anything other than 'sample' the crash goes away.

IEncinas10, 2024-12-29:
> Hi @IEncinas10 , thanks again for your helpful reply. I'm grateful for your support

You're welcome!

> After some debugging I discovered that the code causing the crash resided in a file brought in by my verible.filelist.
>
> My verible.filelist contains thousands of files, so there might be multiple [...]

Thanks for the effort. Indeed, this is what is causing the reported crash (the stack trace is similar). The underlying issue is https://github.com/chipsalliance/verible/issues/2032.

The second snippet seems to be a different problem https://github.com/chipsalliance/verible/blob/2585d6a3b181873e31d5eceaf4116cc3b7c556e4/verible/verilog/analysis/symbol-table.cc#L162 https://github.com/chipsalliance/verible/issues/1946

gobbedy, 2024-12-29:
Thank you for connecting both snippets to an existing issue. Is there any known workaround? Is the only solution to exclude the entire file? (And disable the extension when working within a crash-causing file)

IEncinas10, 2024-12-29:
As far as I know there is no easy fix for the original issue around `sample` other than renaming the problematic functions. Excluding them from `verible.filelist` should also work. Disabling the go-to-definition from the LSP would also work (although I don't know if every editor supports this). 

The second problem seems tractable.

gobbedy, 2024-12-30:
Thank you @IEncinas10 . For now I will just remove verible.filelist altogether.

At first I was going to just exclude the specific files causing the crash, but I've played around with code some more and I'm getting regular crashes in other situations. Tracking each one down would be time consuming.

I also notice that go-to definitions doesn't work for me at all since anytime I 'Go to definition' in anything more than toy code it fails to find anything. One reason is that it appears that it cannot find functions/variables inside packages(?). I suspect that are a variety of causes in a codebase as large/complex as our teams'.

Perhaps I can turn this issue into a feature request to turn off definitions-building? I know you suggested a way to compile from scratch, but my end goal is adoption of Verible by our larger team so that would not be a feasible solution in that context. That would allow us to continue grabbing newer versions of verible without having to recompile every time.

IEncinas10, 2024-12-31:
Sorry to hear that!

I don't know what is the appropriate way to disable definitions-building but we can definitely do it either way or another.

@hzeller do you have any suggestion on how to go about this? Does a command line flag for the language server so that it doesn't advertise certain capabilities look ok to you?