chipsalliance/verible #2315: format one feature at a time (feature request?)

gobbedy, 2024-12-31:
Hello and thanks for your great work. I asked this question in the googlegroup a few days ago but I think github issues may be more active. Sorry for writing in two places, I'll delete if you tell me to.

I work on a verification team with many users and a large code base.

I hope to get the team to adopt a standard formatting style for .sv source code. Once the team agrees on a formatting style, my plan is to verify (using --verify) that all existing .sv files adhere to the agreed upon format, and ask the owners of each violating file to fix.

However, asking users to fix all formatting issues at once would be too disruptive. So I'd like to do this one formatting feature at a time at first, starting with indentation.

Is there a way to format a file on a per-feature basis? For instance, run the formatter with a switch that tells it to only fix the file's indentation.

Basically, what I'd like to be able to do is something equivalent to the linter's "--ruleset none --rules <rules>", but for the formatter

Thanks in advance.

fangism, 2025-01-04:
This is a good question, and has been asked by different people before, so it is worth repeating.

Unlike linting, formatting's features are very much entangled.  The most common formatting constraint is line length, so any individual "rule" like indentation or spacing that may add corrective space runs into the risk of exceeding line length.   You would need to express some form of mutual exclusion among the major classes of formatting rules.  This has not been designed out, but I am open to ideas.   Of course, the long term goal is to reach a level of quality where it is less desirable to pick-and-choose formatting features.

gobbedy, 2025-01-06:
Hi David, thanks for your reply. Should I close this? Since I asked the same question in the googlegroup and we are having an ongoing discussion there.