chipsalliance/verible #2339: formatter doesn't allow named arguments to be on separate lines?

gobbedy, 2025-01-21:
Hello, is there any way to get the formatter to keep the original formatting below? I think I tried every option and nothing worked.

To be clear, we WANT the original formatting (with one argument per line) but the formatter is forcing the actual output, seemingly regardless of what options I set.

Thanks in advance!

**Test case**

```systemverilog
class my_class;

   function void my_func();

      // --------------------------------
      my_func(      .arg1(`TB_AEOH), 
                          .arg2(1),
                          .arg3(1),
                          .arg4(1),
                          .arg5(1),
                          .arg6(1),
                          .arg7(1),
                          .arg8(1),
                          .arg9(1),
                          .arg10(1),
                          .arg11(12),
                          .arg12(8),
                          .arg13(50)
                        );

   endfunction

endclass

```

Include any options or configuration used.

**Actual output**

```systemverilog
class my_class;

  function void my_func();

    // --------------------------------
    my_func(.arg1(`TB_AEOH), .arg2(1), .arg3(1), .arg4(1), .arg5(1), .arg6(1), .arg7(1), .arg8(1),
            .arg9(1), .arg10(1), .arg11(12), .arg12(8), .arg13(50));

  endfunction

endclass
```

hzeller, 2025-01-21:
looks like currently there is not an option yet for the formatter (such as `--named_parameter_indentation=indent`) that applies to function calls not module instantiations.

So for now I suggest to wrap the part of the code in `// verilog_format: off` and `on`

https://github.com/chipsalliance/verible/tree/master/verible/verilog/tools/formatter#disabling-formatting-disable-formatting

fangism, 2025-01-23:
In general, one can force line separations by adding trailing `//` comments (even if empty) to desired lines, but I understand that is not always a reasonable solution.

gobbedy, 2025-01-30:
@fangism thank you, that is a useful workaround to keep in mind