chipsalliance/verible #2346: Error lex/parsing-ing formatted output.

Sladkoeshechka, 2025-01-31:
**Test case**

```systemverilog
class unit_calculator extends unit_base;
  `uvm_component_utils(test::unit_calculator)

  local axi4stream::cfg cfg;
  local axi4stream::item in;
  local axi4stream::item golden;
  local tb_util::calculator sut;

  function new(string name = get_type_name(), uvm_component parent = null);
    super.new(name, parent);
  endfunction

  virtual function void build_phase(uvm_phase phase);
    super.build_phase(phase);
    build_cfg();
    sut = new();
  endfunction

  local function void build_cfg();
    cfg = axi4stream::cfg::type_id::create("cfg");
    assert (cfg.randomize() with {
      bus.omit.id == 1;
      bus.omit.dest == 1;
      bus.omit.user == 1;
      bus.omit.ready == 0;
      bus.omit.keep == 0;
      bus.omit.strobe == 0;
      bus.omit.last == 0;

      bus.width.data == axi4stream::MAX_DATA_WIDTH;
    });
  endfunction

  function void run_tests();
    test_one_mixed_transfer();
    test_two_mixed_transfers();
    test_only_data();
    test_only_position();
    test_only_null();
    test_two_null_transfers();
    test_two_not_full_transfers();
  endfunction

  local function void test_one_mixed_transfer();
    setup_test("one_mixed_transfer");

    initialize_data(in, 1);
    in.get_data(0).set_bytes({
      get_byte(axi4stream::byte_qualifier::DATA, 'h1),
      get_byte(axi4stream::byte_qualifier::POSITION, 'h2),
      get_byte(axi4stream::byte_qualifier::NULL, 'h3),
      get_byte(axi4stream::byte_qualifier::RESERVED, 'h4),
      get_byte(axi4stream::byte_qualifier::DATA, 'h5),
      get_byte(axi4stream::byte_qualifier::DATA, 'h6),
      get_byte(axi4stream::byte_qualifier::DATA, 'h7),
      get_byte(axi4stream::byte_qualifier::DATA, 'h8)
    });

    initialize_data(golden, 1);
    golden.get_data(0).set_bytes({
      get_byte(axi4stream::byte_qualifier::DATA, 'h1),
      get_byte(axi4stream::byte_qualifier::POSITION, 'h2),
      get_byte(axi4stream::byte_qualifier::DATA, 'h5),
      get_byte(axi4stream::byte_qualifier::DATA, 'h6),
      get_byte(axi4stream::byte_qualifier::DATA, 'h7),
      get_byte(axi4stream::byte_qualifier::DATA, 'h8),
      get_byte(axi4stream::byte_qualifier::NULL, 'h3),
[... 476 lines cut ...]
    initialize_data(in, 2);
    in.get_data(0).set_bytes(

    in.get_data(1).set_bytes(

    initialize_data(golden, 1);
    golden.get_data(0).set_bytes(

    assert_sut();
  endfunction
endclass
```

Include any possible diagnostic messages from the formatter.

/home/tsymbalenko-dn/test_verible/unit_calculator.svh: Error lex/parsing-ing formatted output.  Please file a bug.
First error: token: ";" at 59:31:; problematic formatter output is

**Expected or suggested output**

```systemverilog
// This result would look better from the formatter.
```

Citations to published style guides would help.

hzeller, 2025-01-31:
Uh, thanks for the report.

Current formatter is indeed broken for this input. I tried with a very old one and that seems to work fine. Will bisect to see where this got wrong.

hzeller, 2025-02-02:
Alright bi-sected. Looks like this is broken for a while since 3c9419c018d6c91fd7b1c393ad312de26bf237b2 (@mglb FYI).

I currently don't have much time to look into it in detail, but the change looks fairly small, so possibly not too hard to find the issue.

hzeller, 2025-02-02:
This was the corresponding pull request for more context https://github.com/chipsalliance/verible/pull/834