chipsalliance/verible #2350: Installation not successful on MacOS

tang-dynasty, 2025-02-06:
Hi,
does anyone know how can it be solved below issue?

```shell
 fengshi@192  ~  brew install verible 
==> Downloading https://formulae.brew.sh/api/formula.jws.json
==> Downloading https://formulae.brew.sh/api/cask.jws.json
==> Fetching chipsalliance/verible/verible
==> Cloning https://github.com/chipsalliance/verible.git
Updating /Users/fengshi/Library/Caches/Homebrew/verible--git
==> Checking out tag v0.0-3665-ga9394662
HEAD is now at a9394662 Merge pull request #2186 from hzeller/feature-20240527-add-bant
HEAD is now at a9394662 Merge pull request #2186 from hzeller/feature-20240527-add-bant
==> Installing verible from chipsalliance/verible
==> bazel build --jobs=14 --compilation_mode=opt --linkopt=-Wl,-rpath,@loader_path/../lib --verbose_failures --sandbox_debug --macos_sdk_version=
Last 15 lines from /Users/fengshi/Library/Logs/Homebrew/verible/01.bazel:
      |                                              ^~~~~~~~~~~~~
      |                                              absl::index_sequence
external/abseil-cpp~/absl/utility/utility.h:54:12: note: 'absl::index_sequence' declared here
   54 | using std::index_sequence;
      |            ^
external/googletest~/googlemock/src/gmock-internal-utils.cc:221:23: error: use of undeclared identifier 'MakeIndexSequence'
  221 |   return UnBase64Impl(MakeIndexSequence<256>{}, base64);
      |                       ^
external/googletest~/googlemock/src/gmock-internal-utils.cc:220:33: error: no return statement in constexpr function
  220 | constexpr std::array<char, 256> UnBase64(const char* const base64) {
      |                                 ^
3 errors generated.
INFO: Elapsed time: 28.929s, Critical Path: 1.82s
INFO: 3382 processes: 3008 internal, 374 processwrapper-sandbox.
ERROR: Build did NOT complete successfully
```

hzeller, 2025-02-23:
I've updated the [homebrew](https://github.com/chipsalliance/homebrew-verible) formula to the latest version of Verible. Can you try again and see if it works this time ?

hzeller, 2025-03-07:
Did it work ?

tang-dynasty, 2025-03-08:
Hi @hzeller, sorry for replying late, it is working now and installed successfully.
Thanks a lot!

closed 2025-03-08 by tang-dynasty