chipsalliance/verible #2361: prevent merging single-line statements without begin/end onto the same line of if/else

Liuliuliu7, 2025-03-02:
**Test case**

```systemverilog
// Input to the formatter, preferably a reduced test case.
        if (test) 
          test_reg <= 1'b1;
        else
          test_reg <= 1'b0;
```

Include any options or configuration used.
--indentation_spaces=2
--assignment_statement_alignment=align 
--case_items_alignment=align --named_port_alignment=align 
--port_declarations_alignment=align 
--module_net_variable_alignment=align
--wrap_end_else_clauses --line_break_penalty=0 
--compact_indexing_and_selections=false 
--column_limit=80

**Actual output**

```systemverilog
// This doesn't look right.
        if (test) test_reg <= 1'b1;
        else test_reg <= 1'b0;
```

**Expected or suggested output**

```systemverilog
// This result would look better from the formatter.
        if (test) 
          test_reg <= 1'b1;
        else
          test_reg <= 1'b0;
```

I don’t want single-line statements without `begin/end` to be forcibly merged onto the same line as `if/else`. Is there an existing option in `verible-verilog-format` ?

Deutschkiller, 2025-05-08:
there is a workaround method, I tried --column_limit=1 and it worked

Liuliuliu7, 2025-05-08:
It looks nice, thanks

Liuliuliu7, 2025-05-08:
but it has some weird behavior in the module instance, it will put ";" in the next line

kbrunham-intel, 2026-09-22:
@hzeller This looks like a duplicate of #2053. I would suggest closing this issue in favor of that one so we keep a single tracking ticket.