chipsalliance/verible #2366: Comparison of std::string_view iterators from different std::string_view objects is UB

gian21391, 2025-03-20:
The current implementation compares std::string_view iterators that are coming from different std::string_view objects. 
As far as I can understand from [here](https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2009/n2948.html) at "446. Iterator equality between different containers" this is undefined behavior despite both string_views having the same underlying data. This is confirmed when trying to compile the library with MSVC 17.0. 

A clean solution to this is not straightforward but probably using raw pointers encapsulated in a custom range type, rather than string_view would be the easiest solution.

hzeller, 2025-03-26:
Do you have examples where that is happening ? We typically do compare string views (e.g. an identifier found in a file) that are sub strings of a bigger string view (the file content), so comparing these should be fine. So theoretical (not in practice) UB can only arise if we do that on iterators of actually different parts, so would be good to know these places.

We were using raw pointers before, but unfortunately MSVC does not define `std::string_view::iterator` as `const char*` as everyone else, so we had to do some surgery already with `foo.data() + offset` kind of stuff (in #2324).

So while _technically correct_, the choice of MSVC of wrapping their own iterator around `const char*` is causing more headaches than its worth (until #2324 we were using absl::string_view therefore and forcing it to use its own implementation); having said that if there are only a few places where this is causing a warning, we can probably fix that.

If not, we have to switch back to a string_view implementation that works correctly.

gian21391, 2025-03-27:
> Do you have examples where that is happening ? 

This happens every time:

- a distance between iterators of different string_views is calculated. Example [here](https://github.com/chipsalliance/verible/blob/851d3ff4de964bf5ac17b6578dfb2cecb5f89099/verible/common/text/tree-utils.cc#L302) and [here](https://github.com/chipsalliance/verible/blob/851d3ff4de964bf5ac17b6578dfb2cecb5f89099/verible/common/text/token-info.cc#L58);

- a string_view is created from iterators coming from different string_views. Example [here](https://github.com/chipsalliance/verible/blob/851d3ff4de964bf5ac17b6578dfb2cecb5f89099/verible/common/text/tree-utils.cc#L102);

- iterators of different string_views are compared. Example [here](https://github.com/chipsalliance/verible/blob/851d3ff4de964bf5ac17b6578dfb2cecb5f89099/verible/common/util/range.h#L48).

> We typically do compare string views (e.g. an identifier found in a file) that are sub strings of a bigger string view (the file content), so comparing these should be fine. So theoretical (not in [...]

This is indeed a fair assumption considering that std::string_views should be views. However, std::string_view iterators are defined in the same way std::string iterators are: [RandomAccessIterator](https://en.cppreference.com/w/cpp/named_req/RandomAccessIterator). 

> We were using raw pointers before, but unfortunately MSVC does not define `std::string_view::iterator` as `const char*` as everyone else, so we had to do some surgery already with `foo.data() + offset` kind of stuff (in [#2324](https://github.com/chipsalliance/verible/pull/2324)).

This is what I've been doing but I quickly realized that the current assumption is that iterators are pointers and are used as such. An example of this is the decrement of the begin() iterator [here](https://github.com/chipsalliance/verible/blob/851d3ff4de964bf5ac17b6578dfb2cecb5f89099/verible/common/text/tree-utils_test.cc#L874).
So what is the point of using iterators instead of raw pointers?

> So while _technically correct_, the choice of MSVC of wrapping their own iterator around `const char*` is causing more headaches than its worth (until [...]

I think that the choice of using std::string_view iterators with the assumption that are pointers can bite us in the long run since I think that raw pointers are not compliant with what the standard prescribes. 
I think that switching to a string_view implementation that defines iterators as pointers is the way to go.

hzeller, 2025-03-27:
> raw pointers are not compliant with what the standard prescribes.

A pointers is a perfectly correct implementation of a random access iterator. So standard-wise this is not a problem.

gian21391, 2025-03-27:
You are absolutely right. I started from the fact that decrementing foo.begin() is undefined behavior and ended up switching the antecedent and the consequent in my implication. 

> I think that the choice of using std::string_view iterators with the assumption that are pointers can bite us in the long run since I think that raw pointers are not compliant with what the standard prescribes.

Let me reformulate to restore the correctness of this conversation:

I think that the choice of using std::string_view iterators with the assumption that are pointers can bite us in the long run since not all of the operations that are defined for raw pointers are also defined for iterators.

hzeller, 2025-03-27:
what tool do you use to generate the warnings you're seeing ?

gian21391, 2025-03-27:
I started by creating a set of CMakeLists.txt files that are as much as possible mirroring what bazel is doing (since my initial idea was to port verible to my vcpkg registry). I am doing this incrementally and so far I can compile ~60 of the targets defined with cc_test.

I'm compiling the tests with MSVC v143 (Compiler Version 19.43.34809) using ninja. 

The problem actually occurs at runtime. For example the test `TEST(PruneSyntaxTreeAfterOffsetTest, AtEndPrunesNothing)` fails due to `C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\include\__msvc_string_view.hpp(1214) : Assertion failed: cannot subtract incompatible string_view iterators`. 
This is triggered by the std::distance invocation [here](https://github.com/chipsalliance/verible/blob/851d3ff4de964bf5ac17b6578dfb2cecb5f89099/verible/common/text/tree-utils.cc#L302).

This is because the implementation of std::string_view in MSVC includes assertions to check for some of the discussed undefined behaviors when `_ITERATOR_DEBUG_LEVEL >= 1`.

hzeller, 2025-03-27:
So I am considering maybe we should have a `verible::document_view` of sorts, that is like a string view (covers a range of characters), and has semantics of iterators that are ok by comparing iterators if a document_view is a sub-view of a bigger document_view (and we can have `#ifdef`s to possibly even add another field to them for thorough runtime debugging).

These document_views are only compatible with other document views and it would require explicit converstion to a `std::string_view`. That way we have both: a more type-safe way of dealing with the special way verible deals with a document (sub-views all the way down), while also improving the way we can catch bugs. And being compatible with MSVC's string view implementation.

Currently, I will probably not have much time to think about the details, and it might be a few months until I do.
Until then, maybe using a version before #2324 might get you further while you convert the build for you.
(and of course if you want, you can start sketching out a `verible::document_view` situation)

hzeller, 2025-03-27:
(Alternative: set `_ITERATOR_DEBUG_LEVEL` to zero for now  :) ...
the code is well tested including asan so I don't expect (too many) actual bugs in there, but it would help you continue your build conversion).

gian21391, 2025-04-04:
I like a lot the idea of a verible::document_view. I have created an initial version of a possible implementation: https://gist.github.com/gian21391/e8d3f59355dac5cbaf1b8f38b3a74c5a
It is inspired to std::string_view but I have added pointers to the original memory so that is_subview_of and is_same_source can be implemented. I have also wrapped the iterators so that iterator comparisons are valid only if the the containers have the same underlying string. Probably these checks should be enabled only in debug mode (similarly to assert).
I have of course maintained the comparison semantic of std::string_view (comparison of the content) when comparing document_views.

If this goes in the right direction I can open a pull request but I would need some guidance for deciding where this class should go in the folder structure.