chipsalliance/verible #2370: On Windows, CRLF line endings are not preserved when using '--inplace'

cezchitos, 2025-03-25:
On Windows, CRLF line endings are not preserved when using '--inplace'. 
After the formatter is run, the file gets converted to 'LF' (should remain 'CRLF').

hzeller, 2025-03-26:
The https://github.com/chipsalliance/verible/commit/13e0cd0c7f3524cfd49a43ab44e045f431d2c067 should've made it work with Windows at all, I suspect before all kinds of offset might've been wrong, but I also don't have access to a Windows machine, all fixes were informed by roundtrips with the github CI.

I agree there should probably a configuration option that allows to alternatively emit CRLF line endings, which can be set manually and is auto-configured depending on the original input. Please send a pull request.

cezchitos, 2025-04-02:
@IEncinas10 Apologies, I was away for a bit. 

1. I will test this later, but I believe so. I use it without '--inplace' indirectly through TerosHDL and VSCode without any problems, so I'm not sure if that does something in the background. But I will double check. 
2. I just did, and I also tested with the change currently in #2371 ('wb' to 'w') and I've captured the result in a video since it was easier:

https://streamable.com/rvf634

Basically in 'wb' mode (as is on master) CRLF line endings get replaced with LF
In 'w' mode (change in PR), CRLF line endings are preserved, but an extra line is added / removed every time the program runs. 

To note that `verible-verilog-format.exe` is build from master as-is, and `verible-verilog-format-edit.exe` is with the PR change. You can see the line endings in the editor in the bottom right in the video.  

I think the extra line issue in 'w' mode might be related #2383, so maybe there are multiple problems. 

Let me know if you want me to better explain what is happening in the video or if you want me to test something else. 

PS. At the moment after using verible-format on Windows, I just read and write the file in python and that takes care of the line endings for me, just in case someone else also need a (temporary) fix.

cezchitos, 2025-04-02:
I need to check more, but I think the change in #2371 combined with 
```
diff --git a/verible/common/util/file-util.cc b/verible/common/util/file-util.cc
index 27a4c6f1..cebbcd59 100644
--- a/verible/common/util/file-util.cc
+++ b/verible/common/util/file-util.cc
@@ -247,7 +247,7 @@ absl::StatusOr<std::unique_ptr<MemBlock>> GetContentAsMemBlock(

 absl::Status SetContents(std::string_view filename, std::string_view content) {
   VLOG(1) << __FUNCTION__ << ": Writing file: " << filename;
-  FILE *out = fopen(std::string(filename).c_str(), "wb");
+  FILE *out = fopen(std::string(filename).c_str(), "w");
   if (!out) return CreateErrorStatusFromErrno(filename, "can't write.");
   const int64_t expected_write = content.size();
   int64_t total_written = 0;
```
Also fixes #2383, or at least the error on my end that looked like that. And works for both CRLF and LF files (keeping the original formatting), but only on Windows. Should I raise a PR or @IEncinas10 will you look into this as part of your work in #2371?

hzeller, 2025-05-19:
#2371 added an option `--line_terminator=CRLF` that makes the output be `\r\n`; this is now submitted (thanks to @IEncinas10 !).

The functionality does not auto-detect (yet), so for now you have to provide this option to get the desired result @cezchitos

cezchitos, 2025-05-23:
Thank you for your work on this, @IEncinas10 @hzeller. 

I have tried using it with the `line_terminator=CRLF` option and I get the following error:
```
Formatted output is lexically different from the input.    Please file a bug.  Details:
First mismatched token [4]: ("// end of line comment") (#714: "// 32-bit Priority Encoder Module\r") vs. ("// end of line comment") (#714: "// 32-bit Priority Encoder Module")
```

If I use it with the `line_terminator=LF` option, there is no error, but as expected line endings change.

IEncinas10, 2025-05-23:
Thats a shame :( Thanks for testing it! I'll try to trigger it on my side. Could you share a small file that causes the issue?

cezchitos, 2025-05-23:
I have attached a test file. Interestingly, if I remove the ``timescale 1ns / 1ps` line, it works ok. 

[test_file.txt](https://github.com/user-attachments/files/20408656/test_file.txt)

hzeller, 2025-06-05:
Interesting, I think this example in test_file.txt is similar to #2424 where some tokenization in comments is giving troubles (Maybe also closely related #2383 that also had comment related issues)

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

On Linux, stdout and `--inplace` both preserved CRLF in the reduced file.