chipsalliance/verible #2376: Disable linter

qarlosalberto, 2025-03-29:
I would like use the LS in my extension. Currently I offer to the user different linter backend.
Is it possible to disable the linter capabilities and only to have go to definition, hover .. ?

hzeller, 2025-03-29:
Currently it is not possible to switch off the diagnostic directly, but you can disable all linter rules with `--ruleset=none`, which should make that silent (except when it encounters syntax errors).
Would that work ?

qarlosalberto, 2025-03-29:
I would like to disable all, aldo the syntax errors. Some users prefer to use Ventilator or Modelsim as linter.

hzeller, 2025-03-29:
Currently not possible unless you disable this call:
https://github.com/chipsalliance/verible/blob/master/verible/verilog/tools/ls/verilog-language-server.cc#L58

You can add a flag (somewhere around [here](https://github.com/chipsalliance/verible/blob/master/verible/verilog/tools/ls/verilog-language-server.cc#L42)) and use that to switch off adding that listener, then send a pull request.

qarlosalberto, 2025-03-29:
ok, thanks!

hzeller, 2025-03-30:
In #2379 , I've now added that flag that disables auto-pushing the diagnostics. This should solve things.

qarlosalberto, 2025-03-30:
Thank you so much!

qarlosalberto, 2025-03-30:
So how can I disable the linter?

hzeller, 2025-03-30:
Now you only get linting information if your editor requests it; so if you configured your editor to ask a different language server for diagnostics, you get that.

To not get any diagnostics even if asked, set `--ruleset=none` when invoking the verible language server.

Note, many editors allow multiple language servers, so you can have diagnostics enabled for all of them and you get a superset of possibly interesting information.

qarlosalberto, 2025-03-30:
Ok, thanks.

I'm an extension developer, I'm integrating Verible as SV LS and other for VHDL.

hzeller, 2025-04-07:
Did it work ?

qarlosalberto, 2025-04-07:
I'm just trying it out now. Tell me if I'm right:

1. With `--ruleset=none` I disable the style warnings/errors.
2. It doesn't disable syntax errors.

qarlosalberto, 2025-04-07:
I'm using in VSCode:

```
            args: ['--file_list_path', fileListPath, '--ruleset=none'],
```

1. Duplicated messages is fixed.
2. I don't see style warnings/errors.
3. I see syntax errors.

For me it's ok :)

Thanks you very much.

hzeller, 2025-04-07:
And I think the syntax errors you would only see if you instruct VSCode to request diagnostics from that language server. I suspect you have multiple language servers registered and maybe it is possible to selectively tell VSCode which LS to ask for which feature ?

As for `--ruleset=none`, I still recommend maybe to enable some of the lint rules as there are some interesting ones that might be useful https://chipsalliance.github.io/verible/lint.html

qarlosalberto, 2025-04-07:
I see that it's possible to add a function between the client and the server:

```
const clientOptions: LanguageClientOptions = {
    documentSelector: [{ scheme: 'file', language: 'mylang' }],
    middleware: {
        handleDiagnostics: (uri, diagnostics, next) => {
             return undefined;
        }
    }
};
```

So I can discard the diagnostics.

Ok! I will add a basic lint rule file. Thanks!!

qarlosalberto, 2025-04-07:
Do you think that `Enabled by default: true` rules are the more interesting?