chipsalliance/verible #2377: Quiet mode

qarlosalberto, 2025-03-30:
I create the file list automatically from my extension. So the user has multiple projects and when he changes the current project my extension update the file list.

Do I need to restart the server each time that I modify the file list?

I see this message each time that I restart the server:

[image]

Could you have a quiet mode?

hzeller, 2025-04-01:
I am not sure if our implementation currently attempts to re-read the file-list, but it certainly should (I have to look at the exact place). And it should not exit/restart the language server.

Now for the UI representation - I can't do anything about that, every editor does their own thing communicating with the language server. But if you have language server logs (which your editor should provide somewhere), we can better see what is happening.

qarlosalberto, 2025-04-03:
It's curious. I do the follow steps:

1. Create file list with files a,b and c.
2. Update file list with files d and e.
3. Update file list with files a, b and c.

I'm simulating that I change from Project A to B and later to A.  And it seems that Verible add the files. So in 3 Verible is linting a,b,c,d and e. But in my file list I have a,b and c.

If I restart the server between 1-2 and 2-3 it works fine. And in 3 it only lints a,b and c.

hzeller, 2025-04-03:
yes, reloading a file-list is currently not supported. Please file a feature request (and ideally provide a pull request that implements it as I don't have much time these days)

qarlosalberto, 2025-04-04:
I'm not good with C++ but I'm going to investigate the issues. I think that there are 2 different things:

1. It's not possible to reload a file-list.
2. There is a bug when the the server is stopped. I think that it stops abruptly and the client crashes. Because I do the same for RustHDL and Verible LS. With RustHDL it works fine but with Verilbe LS I see:

[image]

Maybe LS needs to send a return message when it's stopped?