chipsalliance/verible #2383: text source file with CRLF do not properly format when using "//" comment

maxicraft0, 2025-04-01:
For example if i have a test.sv file and the file is using CRLF as line end.
If i put some comment using "//", then the formatter will produce extra empty line.
If the file is using LF as line end then it will work fine.

hzeller, 2025-04-01:
is this if you use `--inline` or use the redirect to stdout ?

hzeller, 2025-05-19:
Might this be fixed with your recent change #2371 @IEncinas10 as this looks more closely in parsing and emitting CLRF ?

IEncinas10, 2025-05-19:
I have tried reproducing this but I couldn't. @maxicraft0 could you try if this issue still hapepns with the latest release? Thanks!

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

CRLF file with a `//` comment did not insert an extra blank line.