chipsalliance/verible #2395: goto definition crash with large code base

wpeet, 2025-04-22:
A couple of questions first

 - Your IDE/editor (e.g. vscode, emacs,...) you use with verible LSP ?
   - vscode
 - IDE version:
   - 1.85.2
 - What other SystemVerilog plugins are active alongside ?
   - [Verilog/SystemVerilog Tools](https://marketplace.visualstudio.com/items?itemName=AndrewNolte.vscode-system-verilog)
-----

**What activity failed**
go-to-definition function expected to work
This works with the same file in a smaller code base.
Right mouse click on __neorv32_top__ then __Go To Definition__

**Expectation**
go-to-definition function expected to work
This works with the same file in a smaller code base
**What actually happened**
verible-verilog-ls crashed
**Test case**

If needed, include SystemVerilog sample file(s) that show the code you want to do an operation on
```systemverilog
  // The Core Of The Problem ----------------------------------------------------------------
  // -------------------------------------------------------------------------------------------
  neorv32_top
  #(
    .RVVI_TRACE_EN   (RVVI_TRACE_EN),
    // General
    .CLOCK_FREQUENCY (BBRD_CLK_FREQ)      , // clock frequency of clk_i in Hz
    .HART_ID         (BBRD_HART_ID)    , // hardware thread ID
    .VENDOR_ID       (BBRD_VENDOR_ID)    , // vendor's JEDEC ID
    .CUSTOM_ID       (BBRD_CUSTOM_ID)    , // custom user-defined ID
    .INT_BOOTLOADER_EN(1'b1)           , // boot configuration: true = boot explicit bootloader; false = boot from int/ext (I)MEM

    // On-Chip Debugger (OCD)
    .ON_CHIP_DEBUGGER_EN(1'b1)                   , // implement on-chip debugger
    .OCD_IDCODE_VERSION (4'b0000)                , // version
    .OCD_IDCODE_PARTID  (16'b0000_0000_0000_0000), // part number
    .OCD_IDCODE_MANID   (11'b000_0000_0000)      , // manufacturer id

    // RISC-V CPU Extensions 
    .CPU_EXTENSION_RISCV_A        (1'b0)   , // implement atomic memory operations extension?
    .CPU_EXTENSION_RISCV_B        (1'b0)   , // implement bit-manipulation extension?
    .CPU_EXTENSION_RISCV_C        (1'b1)   , // implement compressed extension?
    .CPU_EXTENSION_RISCV_E        (1'b0)   , // implement embedded RF extension?
    .CPU_EXTENSION_RISCV_M        (1'b1)   , // implement mul/div extension?
    .CPU_EXTENSION_RISCV_U        (1'b0)   , // implement user mode extension?
    .CPU_EXTENSION_RISCV_ZFINX    (1'b1)   , // implement 32-bit floating-point extension (using INT regs!)
    .CPU_EXTENSION_RISCV_ZICNTR   (1'b0)   , // implement base counters?
    .CPU_EXTENSION_RISCV_ZICOND   (1'b0)   , // implement conditional operations extension?
    .CPU_EXTENSION_RISCV_ZIHPM    (1'b0)   , // implement hardware performance monitors?
    .CPU_EXTENSION_RISCV_ZMMUL    (1'b0)   , // implement multiply-only M sub-extension?
    .CPU_EXTENSION_RISCV_ZXFCU    (1'b0)   , // implement custom (instr.) functions unit?

    // Tuning Options
    .FAST_MUL_EN                  (1'b1)   , // use DSPs for M extension's multiplier
    .FAST_SHIFT_EN                (1'b1)   , // use barrel shifter for shift operations
    .CPU_IPB_ENTRIES              (2)      , // entries in instruction prefetch buffer, has to be a power of 2, min 1
    .REGFILE_HW_RST               (1'b1)   , // register file is being reset
    .BOOT_ADDR_INPUT_EN           (1'b1)   , // Enable the boot address input port

    // Physical Memory Protection (PMP)
    .PMP_NUM_REGIONS          (0)      , // number of regions (0..16)
    .PMP_MIN_GRANULARITY      (4)      , // minimal region granularity in bytes, has to be a power of 2, min 4 bytes

    // Hardware Performance Monitors (HPM)
    .HPM_NUM_CNTS             (0)      , // number of implemented HPM counters (0..29)
    .HPM_CNT_WIDTH            (40)     , // total size of HPM counters (0..64)
[... 272 lines cut ...]
    @           0x457be3  verilog::SymbolTable::Build()
    @           0x44ca01  verilog::SymbolTableHandler::BuildProjectSymbolTable()
    @           0x44cb54  verilog::SymbolTableHandler::Prepare()
    @           0x44cbeb  verilog::SymbolTableHandler::FindDefinitionLocation()
    @           0x427652  std::_Function_handler<>::_M_invoke()
    @           0x585210  verible::lsp::JsonRpcDispatcher::CallRequestHandler()
    @           0x585acd  verible::lsp::JsonRpcDispatcher::DispatchMessage()
    @           0x585f29  verible::lsp::MessageStreamSplitter::ProcessContainedMessages()
    @           0x5864e9  verible::lsp::MessageStreamSplitter::ReadInput()
    @           0x5865bd  verible::lsp::MessageStreamSplitter::PullFrom()
    @           0x421e8f  verilog::VerilogLanguageServer::Run()
    @           0x4186cb  main
    @           0x690e0a  __libc_start_call_main
*** SIGABRT received at time=1745329651 on cpu 5 ***
PC: @           0x6c8c9c  (unknown)  __pthread_kill
    @           0x6a8cb0  (unknown)  (unknown)
[Error - 6:47:31 AM] Server process exited with signal SIGABRT.
[Info  - 6:47:31 AM] Connection to server got closed. Server will restart.
true
Verible Verilog Language Server built at v0.0-3946-g851d3ff4
commandline: /prj/sim_data_ICE/verible/v0.0-3946/bin/verible-verilog-ls --rules=-line-length,-no-trailing-spaces,-no-tabs 
I0422 06:47:31.647623   19997 verilog-language-server.cc:240] Initializing with project-root '/prj/pk1332_bbrda_BV/work_libs/peetw/digital/peetw_digital-bbrda0_irv/chips/bbrd/a0'
I0422 06:47:31.647696   19997 symbol-table-handler.cc:109] Found file list under /prj/pk1332_bbrda_BV/work_libs/peetw/digital/peetw_digital-bbrda0_irv/chips/bbrd/a0/verible.filelist
I0422 06:47:31.765415   19997 json-rpc-dispatcher.cc:72] Ignoring notification 'initialized'
```

hzeller, 2025-05-08:
Thanks for the report. Everyone is currently a bit busy, so we're welcoming pull requests :)

nicoklaus16, 2025-07-05:
I've got this same problem with version verible-v0.0-3933-g0e3fe53a and neovim v0.11.2. At the beggining I also thought that it was related to the number of files in `verible.filelist`, but testing that theory I've found otherwise.

Before anything else though, I will note that in my case the failure stack trace looks a little different:
```
F0705 23:44:37.639255   22755 symbol-table.cc:1182] Check failed: 'declaration_type_info_' Must be non-null
"
[ERROR][2025-07-05 23:44:37] ...p/_transport.lua:36     "rpc"   "verible-verilog-ls"    "stderr"        "*** Check failure stack trace: ***
    @           0x5ea80d  absl::lts_20240116::log_internal::LogMessageFatal::~LogMessageFatal()
    @           0x412888  absl::lts_20240116::log_internal::DieBecauseNull()
    @           0x459cf7  verilog::SymbolTable::Builder::EmplaceTypedElementInCurrentScope()
    @           0x45aa3a  verilog::SymbolTable::Builder::Visit()
    @           0x5dc511  verible::TreeContextVisitor::Visit()
    @           0x45a80b  verilog::SymbolTable::Builder::Visit()
    @           0x5dc511  verible::TreeContextVisitor::Visit()
    @           0x45a80b  verilog::SymbolTable::Builder::Visit()
    @           0x5dc511  verible::TreeContextVisitor::Visit()
    @           0x45a80b  verilog::SymbolTable::Builder::Visit()
    @           0x5dc511  verible::TreeContextVisitor::Visit()
    @           0x45a80b  verilog::SymbolTable::Builder::Visit()
    @           0x5dc511  verible::TreeContextVisitor::Visit()
    @           0x45a80b  verilog::SymbolTable::Builder::Visit()
    @           0x5dc511  verible::TreeContextVisitor::Visit()
    @           0x4551f0  verilog::SymbolTable::Builder::DeclareScopedElementAndDescend()
    @           0x45abfd  verilog::SymbolTable::Builder::Visit()
    @           0x5dc511  verible::TreeContextVisitor::Visit()
    @           0x45a80b  verilog::SymbolTable::Builder::Visit()
    @           0x4557ae  verilog::BuildSymbolTable()
    @           0x45589d  verilog::ParseFileAndBuildSymbolTable()
    @           0x455bd3  verilog::SymbolTable::Build()
    @           0x44a9f1  verilog::SymbolTableHandler::BuildProjectSymbolTable()
    @           0x44ab44  verilog::SymbolTableHandler::Prepare()
    @           0x44abdb  verilog::SymbolTableHandler::FindDefinitionLocation()
    @           0x4256c2  std::_Function_handler<>::_M_invoke()
    @           0x585cd0  verible::lsp::JsonRpcDispatcher::CallRequestHandler()
    @           0x58658d  verible::lsp::JsonRpcDispatcher::DispatchMessage()
    @           0x586a03  verible::lsp::MessageStreamSplitter::ProcessContainedMessages()
    @           0x586f79  verible::lsp::MessageStreamSplitter::ReadInput()
    @           0x58704d  verible::lsp::MessageStreamSplitter::PullFrom()
    @           0x41feff  verilog::VerilogLanguageServer::Run()
    @           0x41673b  main
    @           0x69189a  __libc_start_call_main
"
[ERROR][2025-07-05 23:44:37] ...p/_transport.lua:36     "rpc"   "verible-verilog-ls"    "stderr"        "*** SIGABRT received at time=1751755477 on cpu 2 ***
PC: @           0x6c958c  (unknown)  __pthread_kill
```
Now, as I said before, in my case the number of files was not the problem. The issue were a couple of specific files that contained code that was making verible very confused. The way I have been able to find them is by using the debugging variables (NOTE: this will make the log file huge. Your text editor might take a very long time if asked to open the whole thing):
```
VERIBLE_LOGTHRESHOLD=0 VERIBLE_VLOG_DETAIL=3
```
The log now (in my case) has a line that contains the string "EmplaceTypedElementInCurrentScope" just above the failure stack trace. This same line also contains the name of the module (or package or whatever) that is being problematic. I removed the files where the problematic entities were defined from the `verible.filelist` file and as a result the language server does not crash anymore when I use the go-to-definition function.