chipsalliance/verible #2429: Non ANSI-C style input,output port declaration Verilog code linter issue

sscarfi, 2025-06-11:
**Describe the bug**

When applying verible rule `+port-name-suffix` it seems that verible cannot missing `_i`, `_o` when using Non-ANSI style for module ports declaration.

Example:
```systemverilog
module delay_dll_coarse (
    out,
    to_next,
    from_prev,
    in,
    select
);
  output out, to_next;
  input from_prev, in, select;

endmodule
```

**Actual vs. expected behavior**

The tool is not failing.

The correct behaviour would be to see that input/output ports are not following the verible rule `+port-name-suffix`.

hzeller, 2025-06-11:
To fix, expansion of the matching [`verible/verilog/analysis/checkers/port-name-suffix-rule.cc`](https://github.com/chipsalliance/verible/blob/master/verible/verilog/analysis/checkers/port-name-suffix-rule.cc#L101-L124) would be needed. Maybe also some changes needed in `verible/verilog/CST/port.*`.

Do you want to send a pull request ?

hzeller, 2025-06-11:
To see the syntax tree that the linter is operating on, it is useful to get started using `verible-verilog-syntax` with the `--printtree` option

```
bazel-bin/verible/verilog/tools/syntax/verible-verilog-syntax --printtree testfile.sv
```

Also, here are good resources to get started https://github.com/chipsalliance/verible/blob/master/doc/style_lint.md