chipsalliance/verible #2439: Need a flag to quiet syntax errors and do formatting for those files as well!

SunilSegu, 2025-07-21:
Hi,

I was integrating this verible tool in our flow for formatting especially for indentation! When I tried executing verible-verilog-format --inplace <file>. I can see files which has syntax issues are not getting formatted and tool is firing syntax issues.

Can we have a flag to disable syntax issues and do regular formatting for them! I have tried to disable syntax error prints from verilog-format.cc by adding a quiet flag. By this tool stopped firing syntax issues but its not formatting it! Can you please check as its benifical for all sv coders whose code is not required for synthesies!

Example:
File which has this kind of format has syntax issues! 
interface my_interface()

logic signal_1;
`ifdef DEFINED
logic signal_2;
`endif

endinterface

rimanov2000, 2025-08-01:
I see a similar issue. Once Verible hits a syntax error (such as macros defined in open-source packages), it does not format the rest of the code. This is counter intuitive as formatting should not rely on compilation.

I tried disabling syntax checking with `--nopush_diagnostic_notifications` argument. That does remove the red line under the macros but also removes yellow line under code that needs to be formatted (such as, trailing whitespaces or too long a line of code). With this argument formatting is also disabled.

hzeller, 2025-08-04:
Right now, this is unfortunately the way the formatter is implemented: it takes a fully parsed syntax tree and then does formatting on that. What it should do instead is to work on the token stream which allows for a much more forgiving way of dealing with the formatting.

So, it is a known issues, and is on the TODO list to switch to token formatting, but there are also many other projects competing for my attention :) So for now, only files that can be parsed syntax-error free can be formatted.

SunilSegu, 2025-08-04:
Hi,

Thanks for the comment. Is there any ETA scheduled for this?

hzeller, 2025-08-04:
No schedule, sorry. I am busy with other things currently and can't plan for that. Pull requests are accepted, but it will require some detailed knowledge, so will not be done in a day.

There are knowledgeable folks that could implement things on a contract basis if you have a budget...